carla-simulator/carla #6215: VK_ERROR_DEVICE_LOST during long RL training

MoritzNekolla, 2023-02-14:
CARLA version: 0.9.13
Platform/OS: Linux

Error is on server-side
Error-Message:
```
LowLevelFatalError [File:Unknown] [Line: 678]
VulkanRHI::vkDeviceWaitIdle(Device) failed, VkResult=-4
at /home/jenkins/UnrealEngine_4.22/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp:1015
with error VK_ERROR_DEVICE_LOST
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119
Malloc Size=145408 LargeMemoryPoolOffset=276544
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

I run the server with:
```
./CarlaUE4.sh -RenderOffScreen -opengl --carla-world-port=2200
```

This error occurs when training an RL agent on carla Town01_Opt for about a day.
The time of appearance differs from time to time however it always happens.

Notice that I use **openGL** but still receive a **Vulkan** error. Why is that possible?

Does anybody know something about that? So [this ](https://github.com/carla-simulator/carla/issues/2851#issue-617396683) won't work.

MrVanGoghYang, 2023-04-03:
Graphics quality
Vulkan graphics API
Starting from version 0.9.12, CARLA runs on Unreal Engine 4.26 which only supports the Vulkan graphics API. Previous versions of CARLA could be configured to use OpenGL. If you are using a previous version of CARLA, please select the corresponding documentation version in the lower right corner of the screen for more information.

https://carla.readthedocs.io/en/0.9.13/adv_rendering_options/#vulkan-graphics-api

Morphlng, 2023-04-12:
I'm facing the same problem, and I've also noticed that when this happend, `nvidia-smi` will show this info: "failed to initialize NVML: Driver/library version mismatch"