carla-simulator/carla #6216: AttributeError: type object 'CityObjectLabel' has no attribute 'Vehicles' on 0.9.14 version

yaodongC, 2023-02-14:


yaodongC, 2023-02-14:
This happens in 0.9.14 only on both pre-compiled (downloaded from release) and compiled from source CARLA.
The same code works fine on 0.9.12 pre-compiled version.
Only carla.CityObjectLabel.Vehicles cause this problem, other labels such as get_environment_objects(carla.CityObjectLabel.Buildings) works fine in 0.9.14

System: Ubuntu 18
CARLA: 0.9.14 both pre-compiled and built from source

yaodongC, 2023-02-14:
From the surce code, CityObjectLabel.Vehicles has changed to Cars, Trucks and Bus, which are good. But this change in semantic tags has not been reflected in the[ 0.9.14 document](https://carla.readthedocs.io/en/latest/python_api/#carla.CityObjectLabel).

MattRoweEAIF, 2024-04-12:
This is now fixed, thanks for making an issue

closed 2024-04-12 by MattRoweEAIF