carla-simulator/carla #6219: Behavior agent makes collision

Kim-mins, 2023-02-15:
Hi! I'm a beginner to the Carla Simulator, and I got a question while running the simulation.
I'm not expert on using the simulator, so please let me know if there's any missing point.

Here are my settings below:
**CARLA version**: `nightly (installed on 2023.02.14)`
**Platform/OS**: `ubuntu 18.04`
**Problem Statement**:
I wrote my own code for the simulation, and I simply ran the simulation with behavior agent with the following configurations.
- Starting point:
```
Transform(Location(x=239.623291, y=92.175323, z=0.573433), Rotation(pitch=0.000000, yaw=-88.605339, roll=0.000000))
```
- Destination:
```
Transform(Location(x=234.623291, y=-57.824677, z=0.573433), Rotation(pitch=0.000000, yaw=-88.605339, roll=0.000000))
```
When I set the ego vehicle asset as `vehicle.mercedes.coupe`, I could find the collision below (at the end of the video, the ego vehicle collides with a wall on the left side)

https://user-images.githubusercontent.com/50267797/218995267-2e639482-5202-4713-ac00-2dd4260d848d.mp4

For the verification purpose, I also tested from the given example code `automatic_control.py` with small modification on starting point, destination and ego vehicle asset, but the situation was the same.
I wonder if you have any idea on this(a reason of the collision, or something else...).

Thank you for reading this issue and I would really appreciate to you if you tell me any idea.

cdefg, 2023-02-20:
How do you generate the blingbling Xs in the simulator? They are so cool. Do you develop with Unreal Editor or is it a internal api from carla?

Kim-mins, 2023-02-20:
@cdefg They are arrows in 3D. You may enable those by switching `debug` option of the function `run_step` as `True`. In my case,
```python
...
agent = BehaviorAgent(..., behavior='aggressive')
control = self.__agent.run_step(debug=True)
```

glopezdiest, 2023-02-24:
This is really more of a map design issue rather than a behavior agent one. As for why the behavior agent doesn't detect this, agents are designed to only look for vehicle / pedestrians as possible obstacles, so any otehr assets (such as props in the middle of the road) will be ignored.

closed 2023-02-24 by glopezdiest

Kim-mins, 2023-02-28:
@glopezdiest Thank you for the comment! Then I wonder if I should not report any of the collision between behavior agent and vehicles or pedestrians.