carla-simulator/carla #6222: there are some errors during make launch, which may related to BuildCarlaUE4.sh

dujinpeng0249, 2023-02-16:


liuxuzhou, 2023-03-10:
I have the same problem,and I can not solve it too.

suannairen, 2023-05-13:
I have the same problem,and I can not solve it too.

TeamF111, 2023-06-15:
Same problem, did you solve it?

BrellenZhou, 2023-10-22:
Same error. Does anyone have solved this?

MMNavetty, 2024-01-10:
Same here. It happened after launching python script, but only 4/5 of the time (without changing anything and seed is the same). Pretty weird..

romantic-black, 2024-05-17:
${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}

romantic-black, 2024-05-17:
The issue was caused by insufficient memory. After increasing the swap space, the problem was resolved.

enn25, 2024-08-02:
/home/newuser/carla/Util/BuildTools/BuildCarlaUE4.sh: line 209:  2723 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} ${EDITOR_FLAGS}
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 137

does this error is also caused by insufficient memory? By insufficient memory are you talking about the ram or ssd.