carla-simulator/carla #6231: Is it possible to export a custom vehicle from CARLA?

mathiaswold, 2023-02-21:
Hi! 

I have created, imported and set up a custom vehicle in Unreal Engine using the [vehicle content authoring guide](https://carla.readthedocs.io/en/latest/tuto_content_authoring_vehicles/) on Ubuntu 20.04. 

All the relevant files lives in a directory `Kia` in the content folder (next to the `Carla` folder), as seen here:
[image]

This is the content of the `Kia`-directory:
[image]

And it works to spawn it with autopilot in the simulator (don't mind the missing textures):
[image]

Is it possible to create and export a standalone package of this vehicle, so that it can be shared and imported into CARLA later on other computers? I tired `make package ARGS="--packages=Kia"` as described [in the docs](https://carla.readthedocs.io/en/latest/tuto_A_create_standalone/#export-a-package-from-the-ue4-editor), but it errors.

I think the relevant error message is `Error: Package json file not found.`, but I don't see any documentation on how to create this json and where to store it?

[full-make-package-log.txt](https://github.com/carla-simulator/carla/files/10793024/make-package-log.txt)

Thank you for any answers! :)

closed 2023-06-09 by mathiaswold

burnout2k, 2023-10-27:
I have the same problem...
Everything works great in Unreal Editor but how do I export it for a prebuilt version?

sunup494-pixel, 2026-01-14:
I also want to ask this question.