carla-simulator/carla #6234: Issues using Vulkan drivers from `carla` Docker image on Ubuntu 22.04 with NVIDIA GPU

benrbray, 2023-02-22:
# System Information

Basic Information
* Carla Version:  `0.9.14`
* Platform/OS: Ubuntu 22.04
* GPU:  NVIDIA GeForce RTX 3060 Laptop, confirmed to [support Vulkan](https://vulkan.gpuinfo.org/listreports.php?devicename=NVIDIA+GeForce+RTX+3060+Laptop+GPU)

```
$ nvidia-smi
Wed Feb 22 17:03:11 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.85.12    Driver Version: 525.85.12    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   45C    P0    15W /  65W |   1142MiB /  6144MiB |     45%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2205      G   /usr/lib/xorg/Xorg                506MiB |
|    0   N/A  N/A      2667      G   /usr/bin/gnome-shell               92MiB |
|    0   N/A  N/A      3933      G   ...698086980056340607,131072      540MiB |
+-----------------------------------------------------------------------------+

```

```
$ vulkaninfo --summary
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.204

Instance Extensions: count = 19
-------------------------------
VK_EXT_acquire_drm_display             : extension revision 1
VK_EXT_acquire_xlib_display            : extension revision 1
VK_EXT_debug_report                    : extension revision 10
VK_EXT_debug_utils                     : extension revision 2
VK_EXT_direct_mode_display             : extension revision 1
VK_EXT_display_surface_counter         : extension revision 1
VK_KHR_device_group_creation           : extension revision 1
VK_KHR_display                         : extension revision 23
VK_KHR_external_fence_capabilities     : extension revision 1
VK_KHR_external_memory_capabilities    : extension revision 1
VK_KHR_external_semaphore_capabilities : extension revision 1
VK_KHR_get_display_properties2         : extension revision 1
VK_KHR_get_physical_device_properties2 : extension revision 2
VK_KHR_get_surface_capabilities2       : extension revision 1
VK_KHR_surface                         : extension revision 25
VK_KHR_surface_protected_capabilities  : extension revision 1
VK_KHR_wayland_surface                 : extension revision 6
VK_KHR_xcb_surface                     : extension revision 6
VK_KHR_xlib_surface                    : extension revision 6

Instance Layers: count = 3
--------------------------
VK_LAYER_INTEL_nullhw       INTEL NULL HW                1.1.73   version 1
VK_LAYER_MESA_device_select Linux device selection layer 1.3.211  version 1
VK_LAYER_MESA_overlay       Mesa Overlay layer           1.3.211  version 1

[... 166 lines cut ...]
        "implementation_version" : "1",
        "description" : "NVIDIA Optimus layer",
        "functions": {
            "vkGetInstanceProcAddr": "vk_optimusGetInstanceProcAddr",
            "vkGetDeviceProcAddr": "vk_optimusGetDeviceProcAddr"
        },
        "enable_environment": {
            "__NV_PRIME_RENDER_OFFLOAD": "1"
        },
        "disable_environment": {
            "DISABLE_LAYER_NV_OPTIMUS_1": ""
        }
    }
}
```

The `libGLX_nvidia.so.0` library is also available, as I have installed `libnvidia-gl-525`:

```
$ ldconfig -p | grep libGLX_nvidia
	libGLX_nvidia.so.0 (libc6,x86-64) => /lib/x86_64-linux-gnu/libGLX_nvidia.so.0
	libGLX_nvidia.so.0 (libc6) => /lib/i386-linux-gnu/libGLX_nvidia.so.0
```

I've tried a number of different solutions found online, but at this point I'm at a loss for what to try next.  Any help would be appreciated, thank you!

YHao29, 2023-03-07:
I have the same issue and our setup is exactly the same. Have you solved that? thanks

benrbray, 2023-03-07:
> I have the same issue and our setup is exactly the same. Have you solved that? thanks

No, unfortunately I haven't been able to resolve the issue myself yet.  If I do I'll be sure to post the fix here.

YHao29, 2023-03-07:
I just figure it out! you need to modify the ```carla-compose.yml``` as follow:
```sh
version: '3.7'
services:
    simulator:
        image: carlasim/carla:0.9.13
        logging:
          options:
            max-size: "10m"
            max-file: "3"
        privileged: true
        environment:
            SDL_VIDEODRIVER: x11
            DISPLAY: $DISPLAY
        ports:
            - "2000:2000"
            - "2001:2001"
            - "2002:2002"
        command: ./CarlaUE4.sh -quality-level=Low -carla-rpc-port=2000
        volumes:
            - /tmp/.X11-unix:/tmp/.X11-unix:rw
            - /etc/localtime:/etc/localtime
            - /etc/timezone:/etc/timezone
            - /usr/share/vulkan/icd.d:/usr/share/vulkan/icd.d
        restart: always
        deploy:
          resources:
            reservations:
              devices:
                - driver: nvidia
                  count: "all"
                  capabilities: [compute,utility,graphics,display]
``` 
Have a try

bsamadi, 2023-03-07:
> 

I get this error:
```
Status: Downloaded newer image for carlasim/carla:0.9.13
Creating carla_simulator_1 ... done
Attaching to carla_simulator_1
simulator_1  | sh: 1: xdg-user-dir: not found
simulator_1  | sh: 1: xdg-user-dir: not found
carla_simulator_1 exited with code 1
simulator_1  | sh: 1: xdg-user-dir: not found
carla_simulator_1 exited with code 1
```

benrbray, 2023-03-08:
> I just figure it out! you need to modify the `carla-compose.yml` as follow:

Thanks so much, it works!

gabrik, 2023-03-28:
I'm having the same problem, ssh connection to the server, starting the docker compose and Carla just failing without any useful information.
Any suggestion?

doublepoints, 2023-06-05:
@benrbray
I have the same question on can not finda a compitable vulkan driver.
I have tried to modified it, however it not works for me. 
It seems that I have not understand the reason so very. 
Would you please share your yml to me?

xiaosq2000, 2023-12-05:
Hey guys, I had the same problem and somehow managed to solve it. 

My case:
- Carla: 0.9.15
- OS: Ubuntu 22.04.03 with Linux kernel 6.2.0-37-generic
- GPU: NVIDIA GeForce RTX 3070 Ti Laptop with driver 545.29.06
- Docker: 24.0.7
- NVIDIA Container Runtime Hook: 1.14.3
- Docker Compose: 2.21.0

Some additional information which may be helpful:
- My output of `nvidia-smi` and `vulkaninfo` look pretty nice.
- I installed the NVIDIA driver by [Runfile Installers](https://docs.nvidia.com/datacenter/tesla/tesla-installation-notes/index.html#runfile), instead of the package manager on Ubuntu.
- I didn't use the offcial docker image based on "nvidia/vulkan:1.1.121-cuda-10.1--ubuntu18.04", I wrote my own from scratch. Since I'm not sure about the specific minimal working example either now, I decide to post my full [Dockerfile](https://github.com/xiaosq2000/robotics-docker/blob/main/Dockerfile).
- Most importantly I guess, I used a simpler docker-compose file, just "runtime: nvidia" instead of the long "deploy" section. I found that some libraries involving Vulkan were missing when I was trying [YHao29's method](https://github.com/carla-simulator/carla/issues/6234#issuecomment-1458372639).
- Try Carla with the `-nosound` argument.