carla-simulator/carla #6239: carla.Color() do not work correctly when I draw a point using world.debug.draw_point()

NEU-rse, 2023-02-23:


glopezdiest, 2023-02-24:
Could you provide any examples of what is wrong?

HenryHaichaoLiu, 2023-03-18:
> 

I used ```world.debug.draw_point(carla.Location(px, py, 0.3), size=0.025, color=carla.Color(128,0,255,0), life_time=0)``` to show a specified color for the point. But the output of this code is as same as ```world.debug.draw_point(carla.Location(px, py, 0.3), size=0.025, color=carla.Color(255,0,255,0), life_time=0)``` and even ```world.debug.draw_point(carla.Location(px, py, 0.3), size=0.025, color=carla.Color(1,0,1,0), life_time=0)```. 

Thank you for your answer very much!

HenryHaichaoLiu, 2023-03-18:
> Could you provide any examples of what is wrong?

I used world.debug.draw_point(carla.Location(px, py, 0.3), size=0.025, color=carla.Color(128,0,255,0), life_time=0) to show a specified color for the point. But the output of this code is as same as world.debug.draw_point(carla.Location(px, py, 0.3), size=0.025, color=carla.Color(255,0,255,0), life_time=0) and even world.debug.draw_point(carla.Location(px, py, 0.3), size=0.025, color=carla.Color(1,0,1,0), life_time=0).

Thank you for your answer very much!