carla-simulator/carla #6247: how to run Carla 0.9.13 with a specific GPU ??

Dev3apolloInsa, 2023-02-24:
Hello, I have a problem about choosing the GPU when launching the Carla server.

I have a server that contains 4 GPUs (nvidia-smi)(nvidia-smi -L). 

GPU 0: Quadro P4000 (UUID: GPU-22bcf639-2ba6-2a17-e7d4-22828dce08bc)
GPU 1: Quadro P4000 (UUID: GPU-77bd7919-c7e3-9a64-4146-3f592acd27a7)
GPU 2: Quadro P4000 (UUID: GPU-37419e87-d232-ef45-7deb-6885a34c272b)
GPU 3: Quadro P4000 (UUID: GPU-45d63ae7-884e-2b35-f57b-51c77ff43d7f)

I would like to launch several Carla servers (4 exactly) by assigning a specific GPU for each server. But I can't, I saw a command for version 0.9.8 of Carla but it doesn't work with version 0.9.13.
Can you help me find a way to run carla with the desired GPU?? Thanks in advance

Kait0, 2023-02-24:
You can try:
bash path/to/CarlaUE4.sh -graphicsadapter=GPU_ID

or 
SDL_HINT_CUDA_DEVICE=GPU_ID path/to/CarlaUE4.sh

Dev3apolloInsa, 2023-02-24:
I tried both commands but each time carla starts on GPU 0

zycyc, 2024-04-25:
hi thanks for opening this thread.. curious is there any followup on this? getting similar issues except that mine runs on GPU4 regardless of GPU_ID in `bash path/to/CarlaUE4.sh -graphicsadapter=GPU_ID`