carla-simulator/carla #6248: make PythonAPI

arjunsk07, 2023-02-24:
CARLA version: 0.9.13
Platform/OS: windows 10

When I am giving the command make PythonAPI, it is not getting successfull, it is throwing error  in the BuildPythonAPI section saying that it is unable to create process and points out to a location which actually doesnt exist at all.

Some parts of the make process is shown below, the error sections
```
  
**C:\CARLA\carla\LibCarla\source\carla\client\FIleTransfer.cpp(8,10): fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [C:\CARLA\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client.vcxproj]**

**C:\CARLA\carla\LibCarla\source\carla\client\FIleTransfer.cpp(8,10): fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [C:\CARLA\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]**

Cloning into 'C:\CARLA\carla\Build\om2odr-source'...
remote: Enumerating objects: 1006678, done.
remote: Counting objects: 100% (13071/13071), done.
remote: Compressing objects: 100% (1772/1772), done.
remote: Total 1006678 (delta 12401), reused 11299 (delta 11299), pack-reused 993607Receiving objects: 100% (1006678/1006Receiving objects: 100% (1006678/1006678), 717.74 MiB | 10.90 MiB/s, done.

Resolving deltas: 100% (846536/846536), done.
Updating files: 100% (78068/78068), done.
Note: switching to 'ee0c2b9241fef5365a6bc044ac82e6580b8ce936'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at ee0c2b9241 Removed debug warnings
-- Setting build type to 'Release' as none was specified.
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- The C compiler identification is MSVC 19.29.30148.0
-- The CXX compiler identification is MSVC 19.29.30148.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR: C:/CARLA/carla/Build/osm2odr-visualstudio
-- CMAKE_SOURCE_DIR: C:/CARLA/carla/Build/om2odr-source
--
-- Platform:
--     Host: Windows-10.0.19044 AMD64
--     Target: Windows-10.0.19044 AMD64
--     CMake: 3.20.21032501-MSVC_2
--     CMake generator: Visual Studio 16 2019
--     CMake build tool: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe
--     Compiler: MSVC 19.29.30148.0
--
-- Found XercesC: C:\CARLA\carla\Build\\xerces-c-3.2.3-install\lib\xerces-c.lib (found version "3.2.4")
-- Found Proj: C:/CARLA/carla/Build/proj-install/lib/proj.lib
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
-- Enabled features: Windows-10.0.19044 AMD64 MSVC 19.29.30148.0 Release Proj
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
[... 235 lines cut ...]
Unable to create process using 'C:\Users\sd\AppData\Local\Programs\Python\Python39\python.exe setup.py bdist_egg bdist_wheel': The system cannot find the file specified.**

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\CARLA\carla\PythonAPI\carla\dist"!

C:\CARLA\carla>make clean
-[Package]: [Batch params]: --clean --ue-version 4.26
fatal: not a git repository (or any of the parent directories): .git

-[Package]: [ERROR] Carla Version is not set
make: *** [clean] Error 1

```

Here in this I am getting 2 fatal errors:
1. fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [C:\CARLA\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
2. fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [C:\CARLA\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]

And in the last Building Python API for Python 3.
Unable to create process using 'C:\Users\sd\AppData\Local\Programs\Python\Python39\python.exe setup.py bdist_egg bdist_wheel': The system cannot find the file specified.
This error is so strange, because I have only python 3.7 with me and there is no such location like what it mentioned above.
Have checked in environmental variables and everywhere but there is not any python 3.9 related files or paths.

How to get rid of these errors?
@fabianoboril Can you please help me in this?
Thanks in advance

Riha-He, 2023-04-10:
I have the same problem.
Have you solved this problem?

Hassan-BABAOUSMAIL, 2023-04-18:
Chech this issue #2642

NnaYelsel, 2023-05-03:
If you downloaded Carla instead of git cloning it, you will have some problems with Version.h because this one is looking for the version of Carla in git. 

So you need to add a line on Setup.bat.
Remove:  for /f %%i in ('git describe --tags --dirty --always') do set carla_version=%%i 
and add the line:
set carla_version="0.9.<your version>"

saitejeswar1, 2023-11-16:
I am still not able to solve this error. Here is the error log

C:\carla>make PythonAPI
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
        1 file(s) copied.
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\carla\Build\recast-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 12
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "C:\carla\Build\"
0 File(s) copied
0 File(s) copied
        1 file(s) copied.
        1 file(s) copied.

[... 77 lines cut ...]
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-3.7
creating build\lib.win-amd64-3.7\carla
copying source\carla\command.py -> build\lib.win-amd64-3.7\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.7\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.7
creating build\temp.win-amd64-3.7\Release
creating build\temp.win-amd64-3.7\Release\source
creating build\temp.win-amd64-3.7\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\I010910\AppData\Local\Programs\Python\Python37\include -IC:\Users\I010910\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.20348.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\ [... cut ...] LIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
C:\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: 'OSM2ODR.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist"!

dev3225, 2023-12-11:
I am also facing this error with make PythonAPI. 
CMake Error: The source directory "C:/carla/Build/osm2odr-visualstudio/x64" does not exist.
Any updates?

saitejeswar1, 2023-12-12:
@dev3225 check this issue #6898. I could solve it with this method mentioned here.

dev3225, 2023-12-14:
Thankyou @saitejeswar1! it worked now.

haorank, 2024-04-09:
@NnaYelsel I cannot thank you enough！！ it worked！

closed 2025-02-19 by Blyron

CHILLed144, 2025-05-22:
> I am still not able to solve this error. Here is the error log
>
> C:\carla>make PythonAPI -[BuildUE4Plugins]: [Batch params]: --build --all HEAD is now at 260273d Correct descriptor [...]

be careful, the link where to get the assets changed, and is now

`https://carla-assets.s3.us-east-005.backblazeb2.com/`

and the code following it should be chosen upon which version of carla are you interested in
```
https://carla-assets.s3.us-east-005.backblazeb2.com/PUT_FILE_ID_HERE.tar.gz

0.9.5: 20190404_c7b464a
0.9.6: 20190710_0097e66
0.9.7: 20191221_c88604b
0.9.8: 20200306_06b6cb1
0.9.9: 20200422_ea5179a
0.9.10: 20200925_88f9ceb
0.9.11: 20201222_232b876
0.9.12: 20210730_564bcdc
0.9.13: 20211112_d5cfa12
0.9.14: 20221201_5ec9328
0.9.15: 20231108_c5101a5
Latest: 20231108_c5101a5
```