carla-simulator/carla #6250: ModuleNotFoundError: No module named 'carla.Color'

Somdit, 2023-02-25:
CARLA version: 0.9.13
Platform/OS: Windows 11

Hi, I am currently trying to use carla.Color but somehow the submodule was not able to be located although part of the other submodules can be used without problem (e.g., carla.Transform).
I already add the carla paths to my PYTHONPATH and can ensure they are correct. 

Please let me know if you need additional information. Thank you!

elbraa, 2023-03-01:
I have the issue here could you please direct me

Somdit, 2023-03-03:
> 
what i did is i printed the path of the module using print(carla.__file__) and found out the import was not importing the correct carla module but instead importing the mock CARLA module within path "srunner/tests/carla_mocks/". The issue was solved after I manually delete this mock module.

Somdit, 2023-03-03:
> I have the issue here could you please direct me

what i did is i printed the path of the module using print(carla.file) and found out the import was not importing the correct carla module but instead importing the mock CARLA module within path "srunner/tests/carla_mocks/". The issue was solved after I manually delete this mock module.