carla-simulator/carla #6258: Make launch needed or not?

arjunsk07, 2023-02-27:
CARLA version: 0.9.13
Platform/OS: windows 10

I have build the carla simulator on windows from source, I need to know whether I have to give make launch eveytime to open the simulator?

myleskeller, 2023-02-28:
I've found 3 ways to launch the simulator once I've successfully built carla (I'm running 0.9.14 on Win10):

1. `make launch` 
This basically rebuilds LibCarla before opening the UE4 editor.
2. `make launch-only` 
Launches the UE4 editor immediately without rebuilding LibCarla.
3. running `CarlaUE4.exe` found in `<pathToCarla>\Unreal\CarlaUE4\Saved\StagedBuilds\WindowsNoEditor\CarlaUE4\Binaries\Win64\`
Launches standalone mode and immediately starts the simulation (no UE4 editor window).
_Note: this method may only work once you've clicked "Launch" from the UE4 editor at least once._

If you're just trying to immediately start a standalone simulation, 3 is the fastest. If you need the UE4 editor, 2 is the second fastest. if you have modified anything other than the client-side python files, you should probably use 1.

Rumzie01, 2023-03-08:
Adding to this with some general information that I can confirm after my 4 hours compilation decides to stop. 

Unless you are making changes to the carla source code there should be no reason to rebuild using make launch. Once carla is built you should just be able to compile unreal engine itself in Visual Studio for Development Editor x64. And at that you only need to do it if you are modifying your own C++ source code. If you are doing blueprint development there should be no reason to recompile the editor from source again to generate a new binary.

Blyron, 2024-11-08:
You are right. Starting the server pressing play in editor is enough to be able to connect to it through a client. If you are not modifying anything from server or client and you are developing your scripts on top of it, I suggest to download prepackaged Carla version and use it.

closed 2024-11-08 by Blyron