carla-simulator/carla #6268: Using Multi-GPU feature on two computers, not working properly

rezarz98, 2023-02-28:


rezarz98, 2023-02-28:
Hi
I tried to use the new multi-gpu feature on Carla. I have two computers with Nvidia 3060 and 3080 GPUs. The OS for both computers is Ubuntu 22.04 (LTS). I connected them through the network and used this command on the primary server:
**./CarlaUE4.sh -nullrhi**

And I used this for the secondary server:
**./CarlaUE4.sh -carla-rpc-port=4000 -carla-primary-host="primary server's ip" -carla-primary-port=2002 -ini:[/Script/Engine.RendererSettings]:r.GraphicsAdapter=1**

Then I ran the client on the same computer as the primary server. It 
Then it is running and I get a black screen on the secondary server. However, the GPU in non of the computers is utilized. I also tried to use Client examples provided in Python/examples, but it is unable to connect to the server.
 
Why it is not using the GPUs and not running properly, and why I am unable to connect the client to the primary server?

Thanks in advance!

zmz0130, 2023-08-11:
I had the same problem，did you slove it?

rezarz98, 2023-08-11:
@zmz0130 it is best to use their Docker. I got it working that way