carla-simulator/carla #6273: Problem to use make to compile Carla

NnaYelsel, 2023-03-02:
CARLA version: 0.9.13
Platform/OS: Windows
Problem you have experienced: Make doesn't work to compile PythonAPI 

Hi!

I'm trying to compiler Carla and I'm stuck in the Build CARLA step.
```
D:\Unreal_Carla\carla>**make --debug PythonAPI**
GNU Make 3.81
[...]

This program built for i686-pc-msys

Reading makefiles...
Updating goal targets....
 File `PythonAPI' does not exist.
   File `LibCarla' does not exist.
     File `setup' does not exist.
    Must remake target `setup'.
/d/Unreal_Carla/carla/Util/BuildTools\Setup.bat: line 1: @echo: command not found
/d/Unreal_Carla/carla/Util/BuildTools\Setup.bat: line 2: setlocal: command not found
/d/Unreal_Carla/carla/Util/BuildTools\Setup.bat: line 4: rem: command not found
/d/Unreal_Carla/carla/Util/BuildTools\Setup.bat: line 5: syntax error near unexpected token `('
/d/Unreal_Carla/carla/Util/BuildTools\Setup.bat: line 5: `rem rpclib, gtest and boost libraries for CARLA (carla.org).'
```

Several things are weird:
1. The "File XXX does not exist" 
2. The .bat files are not recognized as bat files. 

Even make help does not work:

```
D:\Unreal_Carla\carla>make help
**/bin/sh:** line 0: type: /d/Unreal_Carla/carla/Util/BuildTools\Windows.mk.help: not found
make: *** [help] Error 1
```

Seeing that my code is trying to execute the functions with bin/sh, I also tried to add **SHELL=cmd** option but then, it seem that I have PATH naming problems:

```
D:\Unreal_Carla\carla>make --debug help SHELL=cmd
GNU Make 3.81
...

This program built for i686-pc-msys
Reading makefiles...
Updating goal targets....
 File `help' does not exist.
Must remake target `help'.
'arla' is not recognized as an internal or external command,
operable program or batch file.
make: *** [help] Error 1
```

Does anyone encountered one of these problems and find a solution or a workaround ?

Best regards.

NnaYelsel, 2023-03-06:
Solved it by recompiling cmake.
Sorry for the useless post.

hamdaan-shaikh, 2023-04-30:
I am having the same problem: Make doesn't work to compile PythonAPI, 
how can I recompile Cmake?

kungfusung161, 2024-10-30:
I am running into the same issues with CARLA 0.9.15 and I am using Windows 10 SDK with visual studio 2022. @NnaYelsel can you walk us through how you recompile cmake to resolve this issue please? Thank you!