carla-simulator/carla #6277: How to apply custom animation to walker?

wjsrlahrlco1998, 2023-03-03:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04.05 LTS
---
source code : `carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Walker/WalkerConroller.cpp`
I tried to apply a custom animation to the walker using the following code:
```c++
void AWalkerController::Tick(float DeltaSeconds)
{
  TRACE_CPUPROFILER_EVENT_SCOPE(AWalkerController::Tick);
  Super::Tick(DeltaSeconds);

  float result = 1.0f;

  auto *Character = GetCharacter();
  if (!Character) return;
  /*Original Code*/
  Character->AddMovementInput(Control.Direction,
        Control.Speed / GetMaximumWalkSpeed());

  if (Control.Jump)
  {
    // Character->Jump();

    /*Play AnimMontage Code*/
    UAnimMontage* AnimMontage = LoadObject<UAnimMontage>(NULL, TEXT("/Game/Carla/Static/Pedestrian/Animations/Fallen/AS_general_fallen_Montage"));
    if (AnimMontage != nullptr) {
      // Return 0.0 : Failed
      result = Character->PlayAnimMontage(AnimMontage);
    } else {
      GEngine->AddOnScreenDebugMessage(-1, 15.0f, FColor::Blue, TEXT("Failed to Load AnimMontage"));
    }

    if (result == 0.0f) {
          GEngine->AddOnScreenDebugMessage(-1, 15.0f, FColor::Blue, TEXT("Failed to Play AnimMontage"));
    } else {
      GEngine->AddOnScreenDebugMessage(-1, 15.0f, FColor::Blue, TEXT("Play AnimMontage"));
    }
  }
}
```
This was the original jump code, but I modified it.

When I run this code, no error occurs, but it doesn't apply to the simulator's walker.

How can i solve?
---
**Reference issue**
[1] https://github.com/carla-simulator/carla/issues/4333
[2] https://github.com/carla-simulator/carla/issues/2143
[3] https://github.com/carla-simulator/carla/issues/1615

wjsrlahrlco1998, 2023-03-04:
I solved this problem by modifying the Unreal Engine editor's Animation Blueprint. 

However, I solved it somewhat complicatedly, so if there is a better way, please leave a comment.

thank you.

closed 2023-03-04 by wjsrlahrlco1998

hxri-mz, 2025-02-14:
@wjsrlahrlco1998

Can you please explain how you managed to apply the custom animation?