carla-simulator/carla #6280: Multi GPU pygame window always crashes at launch

ntscheppe, 2023-03-03:
Hey, so im running a dual gpu setup and i want to connect multiple clients to carla. 

Im starting the main server with 
`./CarlaUE4.sh -nullrhi`
and the secondary with
`CarlaUE4.exe -carla-rpc-port=3000 -carla-primary-host=10.12.19.64 -ini:[/Script/Engine.RendererSettings]:r.GraphicsAdapter=1`

I noticed that the secondary server never displays anything (black), but according to this [post ](https://github.com/carla-simulator/carla/issues/6093) that is normal.

so when i now try to run the manual_control.py, the pygame windows just says (Not Responding)

Any ideas on how to fix that?

liyejlu, 2023-03-25:
Hello, I faced the same problem, do you have any idea?

rajesh-kumar, 2023-05-19:
Same error! Any solution please?

Jaker15, 2023-07-18:
I'm finding solution for same error also. anyone plz.

zhouwanqiang-dilu, 2024-07-22:
Same error!