carla-simulator/carla #6288: Running scenarios from scenario runner automatically

arjunsk07, 2023-03-06:
Hello all,

I need to run the example scenarios in scenario runner module automatically, that is without running the manual_control.py file.
So for that in my understanding I think these are the possible ways

1. Using the automatic_control.py
In this when I tried executing this parallely to python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld,
it spawns another vehicle and is executed.

So if I want to execute the scenario with automatic_control.py, I need to load the scenario to automatic_control.py and to have a PID controller?
Is it the right way to do this? or how is the efficient approach to do that?

2. By using the agents
While using the agents, should I look into scenario runner autoagents folder and does it possible to run the scenario alone with the agent.
How can we acheive that?

Also, Is it possible to use agents that are in pythonAPI folder?

Another concern is that, I tried executing the xosc files, and to automatically control that I changed the external_control to simple_vehicle_control but still the vehicle is not moving forward.
what I am missing here?

SExpert12, 2023-06-20:
> Hello all,
>
> I need to run the example scenarios in scenario runner module automatically, that is without running the manual_control.py file. So for that in my understanding I think these are the [...]

[image]

I am getting this error when I run scenario runner files. Can you help me to sort it out.

MichaelBryan2, 2023-06-21:
Hi @SExpert12 

just run `python3 scenario_runner.py --list` and you'll get all available scenarios. The one you're looking for is probably SignalizedJunctionLeftTurn_1.  
`scenario_runner.py --help` is also quite usefull.

SExpert12, 2023-06-21:
Thanks Michael. I am able to run scenario. Can you help with generating Intersection scenario. Actually I wanted scenario_runner for intersection scenario and that is missing in scenario_runner.

Thanks once again for your help.

jmoss7, 2024-05-13:
Hi @arjunsk07 , I am also attempting to create scenarios where agents behaviors are controlled by a script. I'll keep you updated on my progress! I'll be starting by attempting to use automatic_control.py to control a single vehicle, as you seem to be attempting.