carla-simulator/carla #6301: The position of the opendrive map doesn't match the position of it's fbx model when ingesting new map

zhuyingxiang95, 2023-03-10:
hello guys, I have come across a problem when ingesting a new customized 
[image]
map into Carla. After the map loaded into UE4, i have found the position of the imported opendrive map doesn't match its fbx model.  have not found an answer in forum yet. i appreciate it if someone has some comments over it .  (the red path is my opendrive map)

Ahmed-Radwan094, 2023-05-31:
Hi @zhuyingxiang95, did you find a solution to this issue as I am facing it as well.

Ahmed-Radwan094, 2023-05-31:
@zhuyingxiang95 Thank you for your reply. Can you clarify what you mean by local coordinates of the xodr file and where to find it?
Thank you for your help.

zhuyingxiang95, 2023-05-31:
I am not sure if oxdr has a local coordinate. Lanelet 2 format do have.
Have you tried to directly move the fbx in UE4 ?

Ahmed-Radwan094, 2023-06-01:
I tried that, and it works, but I am creating multiple maps and I don't want to manually align each one.
Thank you for your help and advice.

ronyshaji, 2023-08-28:
@zhuyingxiang95 Hey, How did you created a .fbx file? Any opensource software available other than Roadrunner?