carla-simulator/carla #6310: traffic light

Monekyzoon, 2023-03-14:
How to get the coordinates and corresponding colors of all traffic lights in the world?
here is my code
`import carla
import math
import random
import time
import queue
import numpy as np
import cv2

def build_projection_matrix(w, h, fov):
    focal = w / (2.0 * np.tan(fov * np.pi / 360.0))
    K = np.identity(3)
    K[0, 0] = K[1, 1] = focal
    K[0, 2] = w / 2.0
    K[1, 2] = h / 2.0
    return K

def get_image_point(loc, K, w2c):
        # Calculate 2D projection of 3D coordinate

        # Format the input coordinate (loc is a carla.Position object)
        point = np.array([loc.x, loc.y, loc.z, 1])
        # transform to camera coordinates
        point_camera = np.dot(w2c, point)

        # New we must change from UE4's coordinate system to an "standard"
        # (x, y ,z) -> (y, -z, x)
        # and we remove the fourth componebonent also
        point_camera = [point_camera[1], -point_camera[2], point_camera[0]]

        # now project 3D->2D using the camera matrix
        point_img = np.dot(K, point_camera)
        # normalize
        point_img[0] /= point_img[2]
        point_img[1] /= point_img[2]

        return point_img[0:2]

client = carla.Client('10.100.144.17', 4000)
world  = client.get_world()
bp_lib = world.get_blueprint_library()
spawn_points = world.get_map().get_spawn_points()

for i in range(50):
    vehicle_bp = random.choice(bp_lib.filter('vehicle'))
    npc = world.try_spawn_actor(vehicle_bp, random.choice(spawn_points))
    if npc:
        npc.set_autopilot(True)

# spawn vehicle
vehicle_bp = bp_lib.find('vehicle.lincoln.mkz_2020')
vehicle = world.try_spawn_actor(vehicle_bp, random.choice(spawn_points))

# spawn camera
camera_bp = bp_lib.find('sensor.camera.rgb')
camera_init_trans = carla.Transform(carla.Location(z=2))
camera = world.spawn_actor(camera_bp, camera_init_trans, attach_to=vehicle)
vehicle.set_autopilot(True)

# Set up the simulator in synchronous mode
settings = world.get_settings()
settings.synchronous_mode = True # Enables synchronous mode
settings.fixed_delta_seconds = 0.005
world.apply_settings(settings)

# Create a queue to store and retrieve the sensor data
image_queue = queue.Queue()
camera.listen(image_queue.put)

# Get the world to camera matrix
world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

# Get the attributes from the camera
image_w = camera_bp.get_attribute("image_size_x").as_int()
image_h = camera_bp.get_attribute("image_size_y").as_int()
fov = camera_bp.get_attribute("fov").as_float()

# Calculate the camera projection matrix to project from 3D -> 2D
K = build_projection_matrix(image_w, image_h, fov)

# Remember the edge pairs
edges = [[0,1], [1,3], [3,2], [2,0], [0,4], [4,5], [5,1], [5,7], [7,6], [6,4], [6,2], [7,3]]

while True:

    # Retrieve and reshape the image
    world.tick()
    image = image_queue.get()
    frame_path = 'output/%06d' % image.frame

    img = np.reshape(np.copy(image.raw_data), (image.height, image.width, 4))

    # Get the camera matrix
    world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

    bounding_box_set = world.get_level_bbs(carla.CityObjectLabel.TrafficLight)

    for bb in bounding_box_set:
        dist = bb.location.distance(vehicle.get_transform().location)
        if dist < 50:
            forward_vec = vehicle.get_transform().get_forward_vector()
            ray = bb.location - vehicle.get_transform().location

            if forward_vec.dot(ray) > 1:
                p1 = get_image_point(bb.location, K, world_2_camera)
                verts = [v for v in bb.get_world_vertices(carla.Transform())]
                x_max = -10000
                x_min = 10000
                y_max = -10000
                y_min = 10000

                for vert in verts:
                    p = get_image_point(vert, K, world_2_camera)
                    # Find the rightmost vertex
                    if p[0] > x_max:
                        x_max = p[0]
                    # Find the leftmost vertex
                    if p[0] < x_min:
                        x_min = p[0]
                    # Find the highest vertex
                    if p[1] > y_max:
                        y_max = p[1]
                    # Find the lowest  vertex
                    if p[1] < y_min:
                        y_min = p[1]

                height = y_max - y_min
                width = x_max - x_min
                box_size_limit = 3
                if height > box_size_limit and width > box_size_limit:
                    class_num = 0
                    x_center = (x_min + x_max) / 2.0
                    y_center = (y_min + y_max) / 2.0
                    width = x_max - x_min
                    height = y_max - y_min
                    # normal to 0-1
                    x_center /= image.width
                    y_center /= image.height
                    width /= image.width
                    height /= image.height
                    label_debug = '{} {} {} {} {}'.format(class_num, x_center, y_center, width, height)
                    #print(label_debug)
                    cv2.rectangle(img, (int(x_min), int(y_min)), (int(x_max), int(y_max)), (0,0,255, 255), 1)

    #cv2.imwrite(frame_path, img)
    cv2.imshow('ImageWindowName',img)
    if cv2.waitKey(1) == ord('q'):
        break
cv2.destroyAllWindows()
`
I can't get the color of traffic light.

Monekyzoon, 2023-03-14:
import carla
import math
import random
import time
import queue
import numpy as np
import cv2

def build_projection_matrix(w, h, fov):
    focal = w / (2.0 * np.tan(fov * np.pi / 360.0))
    K = np.identity(3)
    K[0, 0] = K[1, 1] = focal
    K[0, 2] = w / 2.0
    K[1, 2] = h / 2.0
    return K

def get_image_point(loc, K, w2c):
        point = np.array([loc.x, loc.y, loc.z, 1])
        point_camera = np.dot(w2c, point)
        point_camera = [point_camera[1], -point_camera[2], point_camera[0]]
        point_img = np.dot(K, point_camera)
        point_img[0] /= point_img[2]
        point_img[1] /= point_img[2]
        return point_img[0:2]

client = carla.Client('10.100.144.17', 4000)
world  = client.get_world()
bp_lib = world.get_blueprint_library()
spawn_points = world.get_map().get_spawn_points()

for i in range(50):
    vehicle_bp = random.choice(bp_lib.filter('vehicle'))
    npc = world.try_spawn_actor(vehicle_bp, random.choice(spawn_points))
    if npc:
        npc.set_autopilot(True)

vehicle_bp = bp_lib.find('vehicle.lincoln.mkz_2020')
vehicle = world.try_spawn_actor(vehicle_bp, random.choice(spawn_points))

camera_bp = bp_lib.find('sensor.camera.rgb')
camera_init_trans = carla.Transform(carla.Location(z=2))
camera = world.spawn_actor(camera_bp, camera_init_trans, attach_to=vehicle)
vehicle.set_autopilot(True)

settings = world.get_settings()
settings.synchronous_mode = True
settings.fixed_delta_seconds = 0.005
world.apply_settings(settings)

image_queue = queue.Queue()
camera.listen(image_queue.put)

world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

image_w = camera_bp.get_attribute("image_size_x").as_int()
image_h = camera_bp.get_attribute("image_size_y").as_int()
fov = camera_bp.get_attribute("fov").as_float()

K = build_projection_matrix(image_w, image_h, fov)

edges = [[0,1], [1,3], [3,2], [2,0], [0,4], [4,5], [5,1], [5,7], [7,6], [6,4], [6,2], [7,3]]

while True:
    world.tick()
    image = image_queue.get()
    frame_path = 'output/%06d' % image.frame

    img = np.reshape(np.copy(image.raw_data), (image.height, image.width, 4))

    world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

    bounding_box_set = world.get_level_bbs(carla.CityObjectLabel.TrafficLight)

    for bb in bounding_box_set:
        dist = bb.location.distance(vehicle.get_transform().location)
        if dist < 50:
            forward_vec = vehicle.get_transform().get_forward_vector()
            ray = bb.location - vehicle.get_transform().location

            if forward_vec.dot(ray) > 1:
                p1 = get_image_point(bb.location, K, world_2_camera)
                verts = [v for v in bb.get_world_vertices(carla.Transform())]
                x_max = -10000
                x_min = 10000
                y_max = -10000
                y_min = 10000

                for vert in verts:
                    p = get_image_point(vert, K, world_2_camera)
                    if p[0] > x_max:
                        x_max = p[0]
                    if p[0] < x_min:
                        x_min = p[0]
                    if p[1] > y_max:
                        y_max = p[1]
                    if p[1] < y_min:
                        y_min = p[1]

                height = y_max - y_min
                width = x_max - x_min
                box_size_limit = 3
                if height > box_size_limit and width > box_size_limit:
                    cv2.rectangle(img, (int(x_min), int(y_min)), (int(x_max), int(y_max)), (0,0,255, 255), 1)

    #cv2.imwrite(frame_path, img)
    cv2.imshow('ImageWindowName',img)
    if cv2.waitKey(1) == ord('q'):
        break
cv2.destroyAllWindows()

Monekyzoon, 2023-03-14:
```
import carla
import math
import random
import time
import queue
import numpy as np
import cv2

def build_projection_matrix(w, h, fov):
    focal = w / (2.0 * np.tan(fov * np.pi / 360.0))
    K = np.identity(3)
    K[0, 0] = K[1, 1] = focal
    K[0, 2] = w / 2.0
    K[1, 2] = h / 2.0
    return K

def get_image_point(loc, K, w2c):
        point = np.array([loc.x, loc.y, loc.z, 1])
        point_camera = np.dot(w2c, point)
        point_camera = [point_camera[1], -point_camera[2], point_camera[0]]
        point_img = np.dot(K, point_camera)
        point_img[0] /= point_img[2]
        point_img[1] /= point_img[2]
        return point_img[0:2]

client = carla.Client('10.100.144.17', 4000)
world  = client.get_world()
bp_lib = world.get_blueprint_library()
spawn_points = world.get_map().get_spawn_points()

for i in range(50):
    vehicle_bp = random.choice(bp_lib.filter('vehicle'))
    npc = world.try_spawn_actor(vehicle_bp, random.choice(spawn_points))
    if npc:
        npc.set_autopilot(True)

vehicle_bp = bp_lib.find('vehicle.lincoln.mkz_2020')
vehicle = world.try_spawn_actor(vehicle_bp, random.choice(spawn_points))

camera_bp = bp_lib.find('sensor.camera.rgb')
camera_init_trans = carla.Transform(carla.Location(z=2))
camera = world.spawn_actor(camera_bp, camera_init_trans, attach_to=vehicle)
vehicle.set_autopilot(True)

settings = world.get_settings()
settings.synchronous_mode = True
settings.fixed_delta_seconds = 0.005
world.apply_settings(settings)

image_queue = queue.Queue()
camera.listen(image_queue.put)

world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

image_w = camera_bp.get_attribute("image_size_x").as_int()
image_h = camera_bp.get_attribute("image_size_y").as_int()
fov = camera_bp.get_attribute("fov").as_float()

K = build_projection_matrix(image_w, image_h, fov)

edges = [[0,1], [1,3], [3,2], [2,0], [0,4], [4,5], [5,1], [5,7], [7,6], [6,4], [6,2], [7,3]]

while True:
    world.tick()
    image = image_queue.get()
    frame_path = 'output/%06d' % image.frame

    img = np.reshape(np.copy(image.raw_data), (image.height, image.width, 4))

    world_2_camera = np.array(camera.get_transform().get_inverse_matrix())

    bounding_box_set = world.get_level_bbs(carla.CityObjectLabel.TrafficLight)

    for bb in bounding_box_set:
        dist = bb.location.distance(vehicle.get_transform().location)
        if dist < 50:
            forward_vec = vehicle.get_transform().get_forward_vector()
            ray = bb.location - vehicle.get_transform().location

            if forward_vec.dot(ray) > 1:
                p1 = get_image_point(bb.location, K, world_2_camera)
                verts = [v for v in bb.get_world_vertices(carla.Transform())]
                x_max = -10000
                x_min = 10000
                y_max = -10000
                y_min = 10000

                for vert in verts:
                    p = get_image_point(vert, K, world_2_camera)
                    if p[0] > x_max:
                        x_max = p[0]
                    if p[0] < x_min:
                        x_min = p[0]
                    if p[1] > y_max:
                        y_max = p[1]
                    if p[1] < y_min:
                        y_min = p[1]

                height = y_max - y_min
                width = x_max - x_min
                box_size_limit = 3
                if height > box_size_limit and width > box_size_limit:
                    cv2.rectangle(img, (int(x_min), int(y_min)), (int(x_max), int(y_max)), (0,0,255, 255), 1)

    #cv2.imwrite(frame_path, img)
    cv2.imshow('ImageWindowName',img)
    if cv2.waitKey(1) == ord('q'):
        break
cv2.destroyAllWindows()
```

Monekyzoon, 2023-03-14:
I successfully got the bounding box of all traffic lights, but I tried many times, but I couldn't get the corresponding traffic light color

Monekyzoon, 2023-03-14:
I have try these funciton:
get_traffic_light
get_traffic_light_from_opendrive_id
get_traffic_lights_from_waypoint
get_traffic_lights_in_junction
world.get_map().get_all_landmarks_of_type('1000001')

But none of the results met expectations

CHNtentes, 2023-11-21:
try carla.TrafficLight.get_state()

Asapling, 2024-02-18:
> 我尝试了以下方法： get_traffic_light get_traffic_light_from_opendrive_id get_traffic_lights_from_waypoint get_traffic_lights_in_junction world.get_map（）.get_all_landmarks_of_type（'1000001'）
> 
> 但结果都没有达到预期

Maybe you can give it a try
~~~python
def control_traffic_light_by_id(world, traffic_light_id, state):
      traffic_light = world.grtactor(traffic_light_id)
      if traffic_light and 'traffic.traffic_light' in traffic_light.type_id:
            traffic_light.set_state(state)

traffic_lights = [actor for actor in world.get_actors() if 'traffic.traffic_light' in actor.type_id]
traffic_light_ids = [traffic_light.id for traffic_light in traffic_lights]
~~~