carla-simulator/carla #6315: Broken images from camera sensors

JonasDoubleU, 2023-03-15:
CARLA version: 0.9.14
Platform/OS: Windows 10
GPU: GTX 1070

When running the tutorial.py file that came with CARLA, the images being saved look like this:

[image]

They also look like that when I try using the RGB sensor in my own code like this
`camera_RGB_bp = blueprint_library.find("sensor.camera.rgb")
camera_transform = carla.Transform(carla.Location(x=1.5, z=2.4))
camera = world.spawn_actor(camera_RGB_bp, camera_transform, attach_to=vehicle, attachment_type=carla.AttachmentType.Rigid)
camera.listen(lambda image: image.save_to_disk('_out/%06d.png' % image.frame, carla.ColorConverter.CityScapesPalette))`

However, when I run the file manual_control.py, that also came with CARLA, and I use the record function, the saved images look like they should, i.e.:

[image]

What can I do to solve this?

mawi42, 2023-03-21:
see https://github.com/carla-simulator/carla/issues/6225

closed 2023-04-12 by JonasDoubleU

TzabarDolev, 2023-09-19:
I wrote a short blog post about this issue on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/16/the-64-division-curse/. And added a short code snippet to fix this.