carla-simulator/carla #6318: make PythonAPI error

zhijiayan, 2023-03-17:
E:\UnrealEngine\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  12
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "E:\UnrealEngine\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "E:\UnrealEngine\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\" --zlib-install-dir "E:\UnrealEngine\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "E:\UnrealEngine\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "E:\UnrealEngine\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "E:\UnrealEngine\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "E:\UnrealEngine\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 12
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "E:\UnrealEngine\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "E:\UnrealEngine\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
已复制         1 个文件。
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "E:\UnrealEngine\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
已复制         1 个文件。
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "E:\UnrealEngine\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
已复制         1 个文件。
-[Setup]: Installing Eigen
    -[install_eigen]: [Batch params]: --build-dir "E:\UnrealEngine\carla\Build\"
复制了 0 个文件
复制了 0 个文件
已复制         1 个文件。
已复制         1 个文件。

    -[install_eigen]: Eigen has been successfully installed in "E:\UnrealEngine\carla\Build\eigen-install"!
    -[install_eigen]: Exiting...
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

[... 481 lines cut ...]
    raise AttributeError(attr)
AttributeError: _ep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "setup.py", line 176, in <module>
    include_package_data=True)
  File "C:\Python37_carla\lib\site-packages\setuptools\__init__.py", line 144, in setup
    _install_setup_requires(attrs)
  File "C:\Python37_carla\lib\site-packages\setuptools\__init__.py", line 133, in _install_setup_requires
    (k, v) for k, v in attrs.items()
  File "C:\Python37_carla\lib\site-packages\setuptools\dist.py", line 443, in __init__
    for ep in pkg_resources.iter_entry_points('distutils.setup_keywords'):
  File "C:\Python37_carla\lib\site-packages\pkg_resources\__init__.py", line 656, in <genexpr>
    for entry in dist.get_entry_map(group).values()
  File "C:\Python37_carla\lib\site-packages\pkg_resources\__init__.py", line 2860, in get_entry_map
    self._get_metadata('entry_points.txt'), self
  File "C:\Python37_carla\lib\site-packages\pkg_resources\__init__.py", line 2531, in parse_map
    raise ValueError("Entry points must be listed in groups")
ValueError: Entry points must be listed in groups

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "E:\UnrealEngine\carla\PythonAPI\carla\dist"!

E:\UnrealEngine\carla>

zhijiayan, 2023-03-17:
I have tried replacing the Python version and pip install – upgrade pip

But the error still exists

Just5D, 2023-05-04:
再运行一遍 make pythonAPI