carla-simulator/carla #6322: SegFault/Unresponsiveness at launch

Imeguras, 2023-03-17:
CARLA version: 0.9.14
Platform/OS: Ubuntu 22.04.2 LTS
Problem you have experienced: The application does not fully launch instantly emitting a segmentation fault or becoming unresponsive
What you expected to happen: The application launching successfully and some kind of graphics to appear on screen
Steps to reproduce: well the segmentation fault is given right at launch so...
Other information (documentation you consulted, workarounds you tried): the specs as recognized by neofetch are, CPU: intel i5-8400(6) 4Ghz, Nvidia GeForce GTX 1060 6GB, Nvidia Tesla K40c. 

I encountered an expired signature when it came to gpg, browsed the keyserver and didnt find a valid up to date one so i decided to install it through the tar manually. It prompted me about a missing dynamic library so i installed it(unfortunatelly i forgot which one) and then i used the command sh CarlaUE4.sh to launch the application unfortunately it just segfaults or prompts me that it isnt responding
The following is a log i grabbed when the latter occurred  
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Killed 
```

carlostojal, 2023-03-17:
Experienced the same problem. 16GB of RAM and it is not the cause. App starts saying it is not responding after a while showing a black screen, when eventually it has a segmentation fault saying it received signal 11, which is SIGSEGV.

misarb, 2023-03-21:
i'm facing the same problem in ubuntu 20.04 with carla 0.9.14 , i try `./CarlaUE4.sh --opengl` but steel the same a black screen showing for some secands.

misarb, 2023-03-21:
just was having the same problem on carla : 0.9.14 on Ubuntu 20.04 ,Looks like a workaround is to force Carla to use nvidia by: DRI_PRIME=1 ./CarlaUE4.sh  and it fixed my issue , oru can use the following command  ./CarlaUE4.sh -prefernvidia

full command 
`DRI_PRIME=1 ./CarlaUE4.sh `
`./CarlaUE4.sh -prefernvidia`

Imeguras, 2023-03-21:
Nope still crashes
To note that one of the cores of the cpu max's out and stays there

misarb, 2023-03-22:
you mention before that u run it on a `CPU: intel i5-8400(6) 4Ghz, Nvidia GeForce GTX 1060 6GB, Nvidia Tesla K40c.` i thing it's a RAM problem which in the documentation they recommend the following :
``` 
An adequate GPU CARLA aims for realistic simulations, so the server needs at least a 6 GB GPU although we would recommend 8 GB. A dedicated GPU is highly recommended for machine learning
```
if it possible try to run old version , in my case i run the `0.9.14` version on a 8GB RAM and i find some difficulty

Anastasiawd, 2024-03-22:
I am facing the same problem...I am using an RTX4090. Did you solve this issue?

Imeguras, 2024-03-22:
> I am facing the same problem...I am using an RTX4090. Did you solve this issue?

Nope although in the following months our team Formula Student Team Lart **might** revisit the sofware. We will let you know if theres news

Imeguras, 2024-03-24:
well @Anastasiawd 

> estou a usar no projeto de licenciatura para gerar datasets sintéticos e está a funfar
> eu não, o meu colega
> em Ubuntu 22.04 com um PC praticamente com as specs do meu
> a causa deve ser alguma cena de versão de drivers da NVIDIA ou assim provavelmente
translated:

im using it(carla) on my bachelors (final) project to generate synthetic datasets and its working
well not me exactly me, my colleague
its working on Ubuntu 22.04 on a computer with pretty much the same specs as mine
it's probably some kind of NVIDIA driver version thing or something

Anastasiawd, 2024-04-11:
> well @Anastasiawd <https://github.com/Anastasiawd>
> estou a usar no projeto de licenciatura para gerar datasets sintéticos e está a funfar
> eu não, o meu colega
> em Ubuntu 22.04 com um PC praticamente [...]

I have resolved this issue. It seems that the problem was caused by the server being unable to properly access the client's display to return images. A specific solution can be attempted by trying the solutions available online for X11 forwarding of server graphical interfaces.

Imeguras, 2024-04-11:
I will close this issue, so its not necro-bumped, either way feel free to refer to this issue in a future date

closed 2024-04-11 by Imeguras