carla-simulator/carla #6326: Carla Server Down

j0hnn, 2023-03-21:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04 focal 
Problem you have experienced: error connecting to carla server
Steps to reproduce: `sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main`
Other information (documentation you consulted, workarounds you tried):

```
Hit:1 https://download.docker.com/linux/ubuntu focal InRelease
Hit:2 https://nvidia.github.io/libnvidia-container/experimental/ubuntu18.04/amd64  InRelease                                           
Get:3 https://nvidia.github.io/libnvidia-container/stable/ubuntu18.04/amd64  InRelease [1,484 B]                                       
Hit:4 https://dl.google.com/linux/chrome/deb stable InRelease                                                                          
Get:5 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]                                                              
Hit:6 http://ppa.launchpad.net/longsleep/golang-backports/ubuntu focal InRelease                                                       
Hit:7 http://in.archive.ubuntu.com/ubuntu focal InRelease                                                                              
Hit:8 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease                                            
Hit:9 http://archive.ubuntu.com/ubuntu focal InRelease                                                                                 
Hit:10 https://brave-browser-apt-release.s3.brave.com stable InRelease                                                                 
Hit:12 http://packages.ros.org/ros2/ubuntu focal InRelease                                                                             
Ign:13 https://s3.amazonaws.com/autonomoustuff-repo focal InRelease                                                               
Get:14 http://security.ubuntu.com/ubuntu focal-security/universe Sources [159 kB]                                                
Hit:15 https://s3.amazonaws.com/autonomoustuff-repo focal Release                                  
Ign:16 https://s3.amazonaws.com/autonomoustuff-repo focal Release.gpg                              
Get:17 http://security.ubuntu.com/ubuntu focal-security/main amd64 DEP-11 Metadata [59.8 kB]
Get:18 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [813 kB]
Hit:11 https://packagecloud.io/github/git-lfs/ubuntu focal InRelease                                          
Get:19 http://security.ubuntu.com/ubuntu focal-security/universe Translation-en [162 kB]
Get:20 http://security.ubuntu.com/ubuntu focal-security/universe amd64 DEP-11 Metadata [95.5 kB]
Err:21 http://dist.carla.org/carla focal InRelease                                                                                     
  Connection failed [IP: 34.227.255.250 80]
Err:22 http://dist.carla.org/carla-0.9.7 all InRelease
  Connection failed [IP: 34.227.255.250 80]
Fetched 1,405 kB in 2min 1s (11.7 kB/s)
Reading package lists... Done
W: Skipping acquire of configured file 'test/source/Sources' as repository 'https://download.docker.com/linux/ubuntu focal InRelease' does not seem to provide it (sources.list entry misspelt?)
W: Failed to fetch http://dist.carla.org/carla/dists/focal/InRelease  Connection failed [IP: 34.227.255.250 80]
W: Failed to fetch http://dist.carla.org/carla-0.9.7/dists/all/InRelease  Connection failed [IP: 34.227.255.250 80]
W: Some index files failed to download. They have been ignored, or old ones used instead.

```

ping to   [IP: 34.227.255.250 80]  returns no response !
may be the server is down ???

robotnetwerk, 2024-03-28:
server still down, nobody cares?

Blyron, 2024-05-24:
We have migrated to backblaze so carla amazon services have been deprecated. 

This is our new server. s3.us-east-005.backblazeb2

Most links will be broken. We suggest to use packaged versions, or use dev branch

closed 2024-05-24 by Blyron