carla-simulator/carla #6332: Failed to connected to dist.carla.org/carla

hengcong, 2023-03-22:
[image]

hengcong, 2023-03-22:
Is there any solution to this error?

tkddnjs98, 2023-03-23:
I got same problem too. It is sad

hengcong, 2023-03-23:
@tkddnjs98 it seems the website was downed

hello-fri-end, 2023-03-25:
It's been like that since the past week! Hope they fix it soon!

Pyrestone, 2023-12-27:
Adding further activity so the stale bot doesn't auto-close it. 

This is still not resolved.

CMakey, 2024-01-02:
still not solved

johndahl, 2024-01-12:
Still an open issue.

robotnetwerk, 2024-03-28:
same here, server down... for a long time, no message on the installation page.. abandonware?

csreid, 2025-02-08:
Remains down. Not a good look! Could we at least get it removed from the docs