carla-simulator/carla #6333: help me plz! RuntimeError: rpc::rpc_error during call in function get_sensor_token

dajiba6, 2023-03-23:
trying to use my client  to connect a carla server on a different machine.

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 4af01bc4 
WARNING: Simulator API version  = 0.9.13 
Traceback (most recent call last):
  File "manual_control2.py", line 1325, in <module>
    main()
  File "manual_control2.py", line 1317, in main
    game_loop(args)
  File "manual_control2.py", line 1191, in game_loop
    sim_world = client.get_world()
RuntimeError: rpc::rpc_error during call in function get_sensor_token

dajiba6, 2023-03-23:
python test_connection.py --host 192.168.0.63
CARLA 0.9.13 connected at 192.168.0.63:2000.

I used this script to test the connection. It seems connected, but when I use other scripts I still get the same error as mentioned above.

lmong11, 2023-03-24:
I face the same problem now........

dajiba6, 2023-03-29:
I download PythonAPI agian. It works now. I used scripts from leaderboard version of carla before. Maybe this is the reason why it failed.

lmong11, 2023-03-29:
I change the version of my python and use .whl to set up my API.	It works now.I guess it's the unmatched version that cause this problem.

Kevin-gkg, 2023-12-11:
Python-3.7 would install Carla-0.9.15 by default, so if you downloaded Carla-0.9.13 and used Python-3.7, then the two version are inconsistent, uninstall and reinstall Carla with:

closed 2024-05-24 by Blyron