carla-simulator/carla #6334: Ureal process has crashed when I run make launch

zhefang-thu, 2023-03-23:
Dear all,

Could you please help me to check this trouble in terms of 'make launch'?

CARLA version: 0.9.12
Platform/OS: ubuntu 18.04
Problem you have experienced: When I run 'make launch', Unreal Engine 4 reported that an Ureal process has crashed: UE4-CarlaUE4, and the progress bar is 71%

```
LoginId:76a6b32683964805973b76da51f5e44c-000003e8
EpicAccountId:

Caught signal 11 Segmentation fault

libUE4Editor-Carla.so!ULoadAssetMaterialsCommandlet::ULoadAssetMaterialsCommandlet() [/usr/Project/OpenCDA/CARLA/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Commandlet/LoadAssetMaterialsCommandlet.cpp:37]
libUE4Editor-CoreUObject.so!UClass::CreateDefaultObject() [/usr/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:3707]
libUE4Editor-CoreUObject.so!ProcessNewlyLoadedUObjects(FName, bool) [/usr/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/UObject/UObjectBase.cpp:996]
libUE4Editor-CoreUObject.so!TBaseStaticDelegateInstance<void (FName, bool), FDefaultDelegateUserPolicy>::ExecuteIfSafe(FName, bool) const [/usr/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:729]
libUE4Editor-Core.so!TMulticastDelegate<void (FName, bool), FDefaultDelegateUserPolicy>::Broadcast(FName, bool) const [/usr/UnrealEngine/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:955]
libUE4Editor-Core.so!FModuleManager::LoadModuleWithFailureReason(FName, EModuleLoadResult&) [/usr/UnrealEngine/Engine/Source/Runtime/Core/Private/Modules/ModuleManager.cpp:516]
libUE4Editor-Projects.so!FModuleDescriptor::LoadModulesForPhase(ELoadingPhase::Type, TArray<FModuleDescriptor, TSizedDefaultAllocator<32> > const&, TMap<FName, EModuleLoadResult, FDefaultSetAllocator, TDefaultMapHashableKeyFuncs<FName, EModuleLoadResult, false> >&) [/usr/UnrealEngine/Engine/Source/Runtime/Projects/Private/ModuleDescriptor.cpp:560]
libUE4Editor-Projects.so!FPluginManager::TryLoadModulesForPlugin(FPlugin const&, ELoadingPhase::Type) const [/usr/UnrealEngine/Engine/Source/Runtime/Projects/Private/PluginManager.cpp:1293]
libUE4Editor-Projects.so!FPluginManager::LoadModulesForEnabledPlugins(ELoadingPhase::Type) [/usr/UnrealEngine/Engine/Source/Runtime/Projects/Private/PluginManager.cpp:1369]
UE4Editor!FEngineLoop::LoadStartupModules() [/usr/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3808]
UE4Editor!FEngineLoop::PreInitPostStartupScreen(char16_t const*) [/usr/UnrealEngine/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:3198]
UE4Editor!GuardedMain(char16_t const*) [/usr/UnrealEngine/Engine/Source/Runtime/Launch/Private/Launch.cpp:127]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/usr/UnrealEngine/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xe6)
UE4Editor!_start()
```

zhefang-thu, 2023-03-23:
terminal error information:
```
1001[2023.03.23-05.42.29:302][  0]LogLinker: Warning: Failed to load '/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset': Can't find file.
1001[2023.03.23-05.42.29:302][  0]LogLinker: Warning: Failed to load '/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset': Can't find file.
[2023.03.23-05.42.29:302][  0]LogUObjectGlobals: Warning: Failed to find object 'Blueprint /Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'
CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'
[2023.03.23-05.42.29:302][  0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'

[2023.03.23-05.42.29:302][  0]Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'

Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
```

zhefang-thu, 2023-03-29:
After running `./Update.sh`, I fixed the above problem.

yuan-mingxi, 2024-02-27:
same problem,but respond "Content is up-to-date." after 

> ./Update.sh

JDIRK3, 2024-05-22:
> same problem,but respond "Content is up-to-date." after
> 
> > ./Update.sh

Did you find any way to fix this issue? I have the same issue.

yuan-mingxi, 2024-05-22:
your email has been received, thank you!邮件已收到，谢谢！

mingyue-gao-hw, 2025-02-04:
> same problem,but respond "Content is up-to-date." after
> 
> > ./Update.sh

Same issue here. Can someone post the solution here? Thanks!

yuan-mingxi, 2025-02-04:
> > > > same problem,but respond "Content is up-to-date." after
> > > > > ./Update.sh
> > > 
> > > 
> > > Same issue here. Can someone post the solution here? Thanks!
> > 
> > 
> > hey！i wrote A brief repo in solving Carla/UE/ubuntu download problem. Check out this link below. Hope it could help you https://blog.csdn.net/yuanerxixi/article/details/136437389?spm=1001.2014.3001.5502
> 
> Hey mingxi, thanks for prompt reply. It's indeed a good reference for how to build carla but did you still have the same problem as in this issue when you built it as in your article? If so, how did you resolve it? I can't find anything about it in the article.
> 
> Besides, what did you mean by "i wrote A brief repo in solving Carla/UE/ubuntu download problem"?

sorry the last time I worked on carla project was one year ago, so I have forgotten most of the details, only remembering that the solution to the problem was to download the virtual machine version again and repeat ’git’ operations( it may also be a problem of network stability. The operation of "reinstalling" maybe a waste of time and may be useless. Please operate with caution.)

yuan-mingxi, 2025-02-04:
> Hey mingxi, thanks for prompt reply. It's indeed a good reference for how to build carla but did you still have the same problem as in this issue when you built it as in your article? If so, how did [...]

My previous graduation project was based on the design of carla platform, so I wrote a record when I built the environment, which is ’ repo’ ，the content in that link.

mingyue-gao-hw, 2025-02-05:
The problem is indeed about inconsistent Content downloaded by the ```Update.sh```. However, if you want to build a specific older version than 0.9.15.2, you might run into another problem where the download link of the content was broken, simply because the download link was changed.

The solution is to make a patch for the ```Update.sh``` of the specific version that you want to build from source, e.g. 0.9.13, according to the content in the ```Update.sh``` of 0.9.15.2.

The broken download link from tag < 0.9.15.2 (e.g. 0.9.13 https://github.com/carla-simulator/carla/blob/0.9.13/Update.sh#L50) is as below:
```
CONTENT_LINK=http://carla-assets.s3.amazonaws.com/${CONTENT_ID}.tar.gz
```

The correct download link (as in https://github.com/carla-simulator/carla/blob/0.9.15.2/Update.sh#L50) is supposed to be below:
```
CONTENT_LINK=https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz
```