carla-simulator/carla #6335: make PythonAPI failed

bbhbbxwy, 2023-03-23:
CARLA version:0.9.11
Platform/OS:windows10
Problem you have experienced:
<details><summary>Details</summary>
<p>

D:\Software\carla>make PythonAPI
D:/Software/carla/Util/BuildTools/Setup.bat: line 1: @echo: command not found
D:/Software/carla/Util/BuildTools/Setup.bat: line 2: setlocal: command not found
D:/Software/carla/Util/BuildTools/Setup.bat: line 4: rem: command not found
D:/Software/carla/Util/BuildTools/Setup.bat: line 5: syntax error near unexpected token `('
D:/Software/carla/Util/BuildTools/Setup.bat: line 5: `rem rpclib, gtest and boost libraries for CARLA (carla.org).'
make: *** [setup] error 258

</p>
</details>

hamdaan-shaikh, 2023-05-01:
same issue here, did you find a solution?

Yehiiaaa, 2023-12-07:
have you found any solution ?