carla-simulator/carla #6340: spawn_actor() error on added props.

raymondlinn, 2023-03-24:
spawn_actor() methods returns RuntimeError

CARLA version: Latest
Platform/OS: Windows 10
Problem you have experienced: added props as described here: "https://carla.readthedocs.io/en/latest/tuto_A_add_props/" and spawn_actor() methods returns RuntimeError
What you expected to happen: should be able to spawn.
Steps to reproduce: 
1) build windows source as described here: "https://carla.readthedocs.io/en/latest/build_windows/"
2) added props as described here in import folder: "https://carla.readthedocs.io/en/latest/tuto_A_add_props/"
3) every thing successful on running "make import"
4) After running "make import" the "Package01" was created under "carla\Unreal\CarlaUE4\Content\Package01"
5) the folder has Config and Static subfolders with under Config, Package01.Package.json and under Static, another Static subfolder which has the added prop folders which contains *.uasset files
6) Run UE editor by "make launch"
7) run "Play"
8) From Jupyter-lab, run these following codes:
import carla
import math
import random
import time

client = carla.Client('localhost', 2000)
client.set_timeout(10.0)
world = client.get_world()

blueprints = [bp for bp in world.get_blueprint_library().filter('*')]
for blueprint in blueprints:
    print(blueprint.id)
    for attr in blueprint:
        print('  - {}'.format(attr))

spawn_points = world.get_map().get_spawn_points()

bp_lib = world.get_blueprint_library()
mybody_bp = bp_lib.find('static.prop.mybody')

# issue on running this line below
mybody= world.spawn_actor(mybody_bp , random.choice(spawn_points))

9) I verified "static.prop.mybody" is in the blueprint library. upon running this above last line gives the RuntimeError: Unknown error while trying to spawn actor.

10) I followed the issues on here: "https://github.com/carla-simulator/carla/issues/4363" but could not make it run.

Thank you,
Raymond

lsangreg, 2023-09-20:
Hi! Have you tried to export a standalone package for carla from what you imported in UE4?

lsangreg, 2023-09-21:
Hi @raymondlinn  have you found a solution to the unkown error when spawning through the python api?

raymondlinn, 2023-09-22:
Hi,

Yes I was able to build it right and spawn the customized actor.

Thanks
Raymond

lsangreg, 2023-09-25:
@raymondlinn  Can you share your solution? I am stucked in the same problem.

Thanks,
Luisa.