carla-simulator/carla #6341: Feature request: orthographic cameras

Kait0, 2023-03-24:
The field of autonomous driving has moved towards bird's eye view (BEV) and 3D perception in recent times e.g: [BEVFormer](https://arxiv.org/abs/2203.17270), [SimpleBEV](https://arxiv.org/abs/2206.07959).

CARLA has camera sensors that automatically can generate labels for semantic segmentation for example.
These are very helpful to generate labels for perspective semantic segmentation.
However they do not work for BEV semantic segmentation as the labels do not create metrically accurate grids due to the perspective projection of the pinhole camera used.

Projects using BEV semantic segmentation like [Roach](https://github.com/zhejz/carla-roach) in CARLA employ complicated workarounds where they re-render the HD-Map as semantic segmentation and render actor bounding boxes into them with pygame.

A much more elegant (and accurate!) solution would be instead to implement an option for [orthographic projection](https://en.wikipedia.org/wiki/Orthographic_projection) in the CARLA camera sensors (or equivalent orthographic cameras).
One could then simply implement place an orthographic camera looking down above the vehicle and obtain metrically accurate BEV labels for free.
One could even get 3D (voxel) perception labels by using multiple orthographic cameras mounted at different heights.

Unreal Engine does support orthographic projection natively, so it should not be too much work to implement this feature.

So in summary, orthographic cameras would allow researchers and practitioners to study and train 3D perception in CARLA.

Kait0, 2023-06-03:
Still think this feature would be valueable for the community.

Spyridonidis98, 2025-02-12:
To this day there is no support for orthographic cameras and actually there is no reason for Carla team to add support for orthographic camera , because you can just make the field of view of the camera smaller and just make the distance of the camera from the ground bigger this will mimic the orthographic effect.
For example lets say that i want a top down view of the ground for a map of 100x100 meters 
i will use a small field of view 'fov' for example 4 degrees and calculate how much distance i need the camera to be from the ground so the camera can see exactly 100 meters from one end to the other.
witch you can use this formula distance_from_ground = (map_size/ 2 * tan(fov/2))
in the image below you can see 2 examples where i use 90 and 4 degress for fov.
In the right image that the fov was set to 4 it has the same effect with using orthographic camera 
[image]

nolan4, 2025-02-12:
2nd'ing @Spyridonidis98's workaround. I've done something similar to produce 2D BEV dynamic occupancy grids: https://github.com/nolan4/carla-2D-dogmas

Kait0, 2025-02-12:
I have tested this workaround already.
Repurposing normal cameras as orthographic cameras has two downsides.
Most importantly that it gets blocked by hills and bridges. 
For example the tunnel in Town 03 would not render the inside of the tunnel but the top of the hill. Similarly such an pseudo orthographic camera would fail if the car is driving below a bridge (and show the top of the bridge instead).
So I cannot use it for my purpose as it is unrealiable unfortunately (want to use BEV semantic as input for a planner).

The second is that I observed CARLA culling vehicles at the edge of the camera sometimes perhaps due to the distance.

Why do you think that Unreal does not support orthographic cameras? The epic documentation claims that Unreal does have an [orthographic camera](https://dev.epicgames.com/documentation/en-us/unreal-engine/orthographic-camera-in-unreal-engine).
If you were mentioning CARLA then yes, the CARLA team has never implemented this feature request since they said they were busy with other features.

Spyridonidis98, 2025-02-12:
Now for use cases where you want to render objects under a certain height like example under trees and bridges. You can use another technique where you basically have the top down view split and rendered by different cameras and then patched together. This will have the same effect since the more cameras you have the smaller the fov of each camera. Like shown in the image below.

[image]

Using this technique you can render using what ever height you want, of course the closer you are to the ground the more cameras you have to use to achieve a better orthographic effect. Below is an example of using different amount of cameras using a distance of 5 meters from the ground where the trees dont obscure the cameras like in the previous example. 

[image]

Of course i will state the obvious that  the more cameras you add to create a better orthographic effect the slower your simulation, even if you keep the total resolution added from all the cameras constant. At 64 cameras at my machine it turns into a slide show. So you can practically only use that technique to gather data rather than running live.  
Also a small note for the technique to work properly you have to run carla in synchronous mode, that way when you try to get data from the many cameras you can be sure that the data belongs to the same frame.   

Now the only thing that we can do is wait until the carla teams adds orthographic camera support.