carla-simulator/carla #6344: Using carla as a MQTT client (send and subscribe to MQTT messages)

svenwinkelmann, 2023-03-26:
Hi!

I think about using Node-Red to implement logic in a low-code manner between different modules, including carla. The idea is to have a MQTT broker to which every module connects (carla, machine learning modules, rule-based dialog models, ..) and define the logic in Node-Red. This means I need to send events from carla with MQTT messages to the broker and subscribe to MQTT messages. 

Is there by chance any MQTT implementation for carla available? It might also work with a REST or http interface. Does carla provide such an interface?

Any help would be appreciated! I am also grateful for any hints pointing me in the right direction to implement such an interface (and for sure provide it to the community!).

Thanks,
Sven

lym-ay, 2023-03-27:
python has mqtt api of phao-mqtt.You can use mqtt in carla's code to receive data, process it, and send the results using mqtt

svenwinkelmann, 2023-03-31:
Thanks, so there is nothing out of the box? Phao-mqtt I know. My first approach would be to write a python script, setting the carla server in sync mode and sending/receiving MQTT events with each tick. Does this make sense? Or would you recommend any other solution?

gabrik, 2023-04-19:
Hi @svenwinkelmann, we have a dataflow python-based pipeline that gets the data out of Carla and processes it to produce a ground-truth-based driving, you can find it here: https://github.com/ZettaScaleLabs/STUNT and a little sneak peek here: https://zenoh.io/blog/2023-02-10-zenoh-flow/#example-applications

It gets the data out from the simulator using [Eclipse Zenoh](https://github.com/eclipse-zenoh/zenoh) and the processing pipeline runs on [Zenoh-Flow](https://github.com/eclipse-zenoh/zenoh-flow).

The dataflow approach is exactly what NodeRed provides, but we are missing a graphical interface to define the graph.
Instead, it is done via a YAML file that describes the interconnections, in the case of ground-truth-based driving this is the file: https://github.com/ZettaScaleLabs/STUNT/blob/main/graphs/drive/zenoh-perfect-perception-drive.yml

The script getting data from the simulator is: https://github.com/ZettaScaleLabs/STUNT/blob/main/utils/zenoh-sensors.py and then this data is processed by the aforementioned pipeline.

If you want to know more feel free to reach out on our Discord server: https://discord.com/invite/2GJ958VuHs