carla-simulator/carla #6362: ㅇㅇ

seungtae-Han, 2023-03-30:
ㅇㅇ

moenasrallah, 2023-03-30:
Error is because you are using stock unreal engine 4.26.
You need carla's
https://github.com/CarlaUnreal/UnrealEngine

hlwei, 2023-04-18:
I have the same problem. Anyone has fixed it? or is there any solution for this? Thx

endris1011, 2023-04-22:
I have the same problem

moenasrallah, 2023-04-22:
@hlwei & @endris1011 , read my comment right on top of yours. That's the solution.

moenasrallah, 2023-04-22:
@seungtae-Han can you close this issue now?

endris1011, 2023-04-22:
> @hlwei & @endris1011 , read my comment right on top of yours. That's the solution.

So I need a „carla-version“ of Unreal Engine? Is it similar to UE 4.26?

closed 2023-04-24 by seungtae-Han

b1llow99, 2023-08-17:
Is the problem solved? I have the same problem, I hope someone can tell me the answer

seungtae-Han, 2023-08-18:
Hi, @hlwei @endris1011 @b1llow99 .
I'm sorry for late reply.
Do not set branch to '4.26' in Carla/UnrealEngine github.

I set branch to 4.26 in image of my first question, but there is no problem if i set it to carla like the picture below.
@moenasrallah 's answer was correct.

[image]

SJ824, 2023-09-11:
I have also same problem. I already installed Unreal engine 4.26 release version, and install the carla. Do I need to delete and reinstall them all?

seungtae-Han, 2023-09-11:
> I have also same problem. I already installed Unreal engine 4.26 release version, and install the carla. Do I need to delete and reinstall them all?

Would you like to delete the Unreal Engine that is currently installed and install it again using the above method?

SJ824, 2023-09-11:
혹시 한국 분 이신거 같아서, 한글로도 남겨봅니다. 현재 설치되어 있는 unreal engine과 carla를 모두 삭제하고 위의 carla unreal engine으로 다시 설치하고 그 위에 carla를 install하게 되면 위의 오류 없이 정상적으로 설치가 가능한가요?

If I delete both my current installation of unreal engine and carla and reinstall with the above carla unreal engine and install carla on top of it, will it install normally without the above error?

seungtae-Han, 2023-09-11:
@SJ824 
Unreal Engine만 재설치하시고, UE path만 잘 설정하면 될 것이라고 예상됩니다.
제가 시행착오를 겪으면서 정리한 설치 방법을 아래에 추가합니다.
따라해보시고, 문제가 있으시면 말씀해주세요.
It is expected that you only need to reinstall the Unreal Engine and set up the UE path well.
I'm adding the installation method that I have documented through trial and error below.
Please follow the instructions below, and if you encounter any issues, let me know.

### **How to Install UE4 for carla and CARLA**
My environment
- CARLA: 0.9.14
- Ubuntu: 20.04
- GPU: NVIDIA GeForce RTX 3090
- Python: 3.8

1. Install the required software
1-1) Create activate the conda environment
`conda create -n [ENVNAME] python=3.8`
`conda activate [ENVNAME]`
1-2) Install
`sudo apt-get update &&
sudo apt-get install wget software-properties-common &&
sudo add-apt-repository ppa:ubuntu-toolchain-r/test &&
wget -O - https://apt.llvm.org/llvm-snapshot.gpg.key|sudo apt-key add - &&
sudo apt-add-repository "deb http://apt.llvm.org/xenial/ llvm-toolchain-xenial-8 main" &&
sudo apt-get update`
1-3) If Ubuntu version is 20.04, enter the command below (see carla document for other versions)
`sudo apt-add-repository "deb http://apt.llvm.org/focal/ llvm-toolchain-focal main"`
`sudo apt-get install build-essential clang-10 lld-10 g++-7 cmake ninja-build libvulkan1 python python-dev python3-dev python3-pip libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev git`
`sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-10/bin/clang++ 180 &&sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-10/bin/clang 180`

2. Install the Python dependencies
2-1) Check that the pip version is 20.3 or higher
`pip3 -V` or `pip -V`
If pip version is less than 20.3, upgrade the pip.
`pip3 install -upgrade pip`
`pip install -upgrade pip`
2-2) Install
`pip install --user setuptools &&pip3 install --user -Iv setuptools==47.3.1 &&pip install --user distro &&pip3 install --user distro &&pip install --user wheel &&pip3 install --user wheel auditwheel`

3. Install Unreal Engine for carla
3-1) Install carla source code from https://github.com/CarlaUnreal/UnrealEngine
**Do not set branch to '4.26' in Carla/UnrealEngine github.**
[image]
3-2) Unzip the 'UnrealEngine-carla.zip' and excute the following command.
`cd UnrealEngine-4.26`
`./Setup.sh && ./GenerateProjectFiles.sh && make`
3-3) Verify that Unreal Engine is successfully installed.
`cd ~/UnrealEngine-carla/Engine/Binaries/Linux && ./UE4Editor`

4. Install the CARLA
4-1) Install aria2 library.
`sudo apt-get install aria2`
4-2) Install
`cd ~ && git clone https://github.com/carla-simulator/carla`
`cd ~/carla && git checkout -b tags/0.9.14`
`./Update.sh`

5. Build the CARLA
5-1) Modify bashrc Files
`cd && gedit ~/.bashrc`
Add text to bottom line: export UE4_ROOT=~/ UnrealEngine-carla
`source ~/.bashrc`
5-2) Enter 'UE4_ROOT=/'PATH'/UnrealEngine-carla' on the 6th line of 'carla/Util/BuildTools/BuildCarlaUE4.sh'.
5-3) Compile the Python API Client
`cd ~/carla && make PythonAPI` (Build successful only when the text 'BuildPythonAPI.sh : Success!' is printed)

6. `make launch`

SJ824, 2023-09-13:
@seungtae-Han 
위 방법대로 진행하니, unreal engine editor에서 환경 빌드는 가능하였는데, ./CarlaUE4.sh 파일이 없어 환경을 실행이 안됩니다. 이 문제는 어떻게 해결하셨나요?
환경을 실행하는 방법을 모르겠습니다..

seungtae-Han, 2023-10-04:
> @seungtae-Han 위 방법대로 진행하니, unreal engine editor에서 환경 빌드는 가능하였는데, ./CarlaUE4.sh 파일이 없어 환경을 실행이 안됩니다. 이 문제는 어떻게 해결하셨나요? 환경을 실행하는 방법을 모르겠습니다..

@SJ824 
답변이 늦어져서 죄송합니다..
Carla 설치 방법은 2가지로 나뉘는데, 첫번째는 source로 부터 설치하는 방법이고, 두번째 방법은 package로 부터 설치하는 방법입니다.
제가 정리한 방법은 source로 부터 설치하는 방법이고, 이 방법으로 설치하면 CarlaUE4.sh 파일이 존재하지 않기 때문에 다음 방법으로 실행해야합니다:
(1) 'make launch' 명령
(2) Unreal engine 상에서 map 선택
(3) Play 버튼 클릭
(4) PythonAPI 코드 실행

두번째 방법으로는 시도한 적 없습니다. 이 이슈를 참고하면 좋을 듯 합니다:
https://github.com/carla-simulator/carla/issues/22

첫번째 방법으로 설치 후 실행하는 것에 대해 궁금하시다면 말씀해주세요. 더 자세하게 설명드리겠습니다.

Carla installation methods can be divided into two: the first is installing from the source, and the second is installing from a package. The method I summarized is for installing from the source, and when you install this way, there won't be a CarlaUE4.sh file. You should follow the next steps to run it:
(1) Use the 'make launch' command.
(2) Select a map within the Unreal Engine.
(3) Click the "Play" button.
(4) Execute your Python API code.

I haven't attempted the second method. You may find more information in this issue:
https://github.com/carla-simulator/carla/issues/22

If you have questions about how to install and run it using the first method, please feel free to ask, and I will provide more detailed instructions.

seungtae-Han, 2023-10-05:
@Dragonking3x 
I don't know how you installed Unreal Engine, and I don't know which version of Unreal Engine you installed.
Also, I'm not sure if both versions(0.9.13 and 0.9.14) of Carla can be built in the same environment.

I was able to install it successfully as I described above.

moenasrallah, 2023-10-07:
@Dragonking3x and whoever else still has this issue

It IS a matter of following the instructions incorrectly. You can't build Carla with stock unreal. You cant have both stock unreal and carla's modified one on the repo and branch they have in the instructions. You have to just have theirs and MAKE SURE that you are BUILDING carla with that version. Not building it with stock. How are you verifying your versions and compatibility?

Carla devs needed to modify the unreal engine to enable some features in 0.9.14.
I advise against just downloading the engine as a zip. It will cause confusion or make it hard to correct things.

**Clone it, make sure youre on the carla branch, make sure that you build the engine again after, and rebuild 0.9.14 carla. It will work.**

This ticket is titled "Error executing 'make launch', 'Renderer/Public/GBufferView.h'". The GBufferView header cannot be compiled because its not FOUND IN STOCK UNREAL ENGINE. Its something added to CARLA's branch. If you do a git diff between both versions of stock Unreal and Carla's, you will see that there are only a few changes and its the GBuffer view being added.

I cannot be more clear than this. If this doesn't help you, I'm unsure what other answer will help you. The author of this issue even fixed it.

and @Dragonking3x  if you have 0.9.13 working and not 0.9.14 working, then thats a clear indicator that you are no on the right unreal editor version. You should not need to "fix some bugs" when its working for others. You could be causing more underlying issues. You need to provide more information than just things not working. Why not post screenshots? If you need more support, take it to the discord server and ask users there: https://discord.com/invite/8kqACuC

kkt474, 2024-08-30:
> I have the same problem. Anyone has fixed it? or is there any solution for this? Thx

> Error is because you are using stock unreal engine 4.26. You need carla's https://github.com/CarlaUnreal/UnrealEngine
> 
> make sure its the carla branch
> 
> edit: not sure what the downvotes are for. This is the correct answer. Just follow instructions closely

yes, this is the right answer.

hitmaster999, 2024-10-15:
> [CarlaUnreal/UnrealEngine.git](https://github.com/CarlaUnreal/UnrealEngine.git)

I want to download this branch（https://github.com/CarlaUnreal/UnrealEngine）, but a 404 error occurs after accessing it. See the picture as follows. Do you know how to access it correctly?
[image]