carla-simulator/carla #6370: Bug: Instance segmentation SceneProxy does not work with Ray Tracing

Domsall, 2023-03-31:
I could not get Unreals ray tracing (Post-Processing) to work with Carla.

After searching for days I found the reason: The SceneProxys in TaggedComponent.cpp add the instance segmentation material to the objects and after that rays are not bounced back from the objects anymore. This unfortunetaly also happens in RGB-Mode, not only in instance segmentation mode.

Hack for now: Outcomment the Code in the functions `FTaggedStaticMeshSceneProxy`, `FTaggedSkeletalMeshSceneProxy`, `FTaggedInstancedStaticMeshSceneProxy`, `FTaggedHierarchicalStaticMeshSceneProxy`.
Then Ray-Tracing works! But the instance segmentation camera does not work anymore.

Do you know how to change the Code, that both effects work at the same time?

Domsall, 2023-06-01:
Can a developer fix this or knows how to?

Blyron, 2025-04-01:
This is working on CARLA 0.10.0. We are not supporting raytracing on 0.9.X

closed 2025-04-01 by Blyron