carla-simulator/carla #6376: Carla server breaking off (How to use OpenGL graphic in Carla 0.9.12+)

dk-kling, 2023-04-03:
CARLA version: 0.9.12+ (I tried 0.9.12, 0.9.13, 0.9.14)
Platform/OS: Ubuntu 18.04 (I also tried docker)
Problem you have experienced: Carla server, unreal engine shuts down during iteration of driving scenario.
What you expected to happen: No break off
Steps to reproduce: load world -> run scenario -> destroy all actor + reload world -> iteration of run scenario
Other information (documentation you consulted, workarounds you tried):

Hi there!
When performing the above execution, the following error message is obtained through Carla Unreal Engine.

```
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Although Carla [officially supports](https://carla.org/2021/08/02/release-0.9.12/) only Vulkan graphics since version 0.9.12, is it still possible to use OpenGL to run Carla?

The reason why I need to use OpenGL is that running the simulator multiple times with Vulkan can cause some issues due to the way Vulkan handles the graphics memory. On the other hand, OpenGL is more stable in this regard and allows you to run the simulator multiple times without any problems. And the reason why I need to use Carla 0.9.13+ is that I have to use the function only supported on 0.9.13+.

Any advice on other ways to fix this in 0.9.13 and later versions is also welcome!
Thanks in advance!

- Version prior to 0.9.12
0.9.9.4: https://github.com/carla-simulator/carla/issues/5216
0.9.10.1: https://github.com/carla-simulator/carla/issues/3618
0.9.11: https://github.com/carla-simulator/carla/issues/3903

- Similar Issue
https://github.com/carla-simulator/carla/issues/5918 
https://github.com/carla-simulator/carla/issues/6215 
https://github.com/carla-simulator/carla/issues/6355

Youngzipper, 2023-09-30:
I encountered the same problem. Have you got a solution yet?

Youngzipper, 2023-10-05:
> Having the same issue, but no solution has been found yet ...

Have you got a solution? I have the same problem

dk-kling, 2023-10-05:
I haven't found a solution yet. 
I am restarting the Carla Simulator after each run in order to address this problem.
If you find a solution, please leave a comment on this issue.

Youngzipper, 2023-10-05:
> I haven't found a solution yet. I am restarting the Carla Simulator after each run in order to address this problem. If you find a better solution, please leave a comment on this issue.

I even can not restart CARLA. I use "subprocess.Popen(carla_path)" in order to restart CARLA when it crashes, but it does not function. I can not solve this problem either. 
How do you restart CARLA? Use python script or command in the terminal?

Victornovikov, 2024-01-28:
Same issue?
Can someone explain what is going on here? I am using NVIDIA + ubuntu 22.04.
Is there any way to solve it?

```
/CarlaUE4.sh                  
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
amdgpu: The CS has been rejected, see dmesg for more information (-2).
LowLevelFatalError [File:Unknown] [Line: 803] 
VulkanRHI::vkQueueSubmit(Queue, 1, &SubmitInfo, Fence->GetHandle()) failed, VkResult=-4
 at /home/jenkins/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanQueue.cpp:71 
 with error VK_ERROR_DEVICE_LOST
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

```

IAMHAADICOOL, 2025-04-21:
I am also facing the same issue, any solutions?