carla-simulator/carla #6381: Please lift Carla to a more recent Ubuntu Version, e.g. 22.04

buechmarc, 2023-04-04:
Dear Carla - Team, we have great trouble compiling carla. We tried Ubuntu 20.04 and 18.04, and each time we run into problems. Either by using tricks to use an older llvm or simply that your scripts cannot download outdated libraries, for example,

libecerxes 3.2.3 does not even exist anymore under the provided download link. 

------
 https://ftp.cixug.es/apache//xerces/c/3/sources/xerces-c-3.2.3.tar.gz
Auflösen des Hostnamens ftp.cixug.es (ftp.cixug.es) … 193.144.61.75
Verbindungsaufbau zu ftp.cixug.es (ftp.cixug.es)|193.144.61.75|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2023-04-04 14:19:28 FEHLER 404: Not Found.

qhaas, 2023-05-04:
Ubuntu 18.04 reaches [end of mainstream support this month](https://ubuntu.com/18-04), recommend dropping support for it in the next release to save QA time.

dburakovtechsolutions, 2023-05-24:
We have managed to build CARLA under Ubuntu 22.04 with slight modifications to the official [tutorial](https://carla.readthedocs.io/en/latest/build_linux/). Modifications are as follows:

Install the following dependencies:
```
sudo apt-get install -y build-essential clang-12 lld-12 cmake ninja-build libvulkan1 libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev git aria2 libstdc++-12-dev  libbz2-dev
sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-12/bin/clang++ 180
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-12/bin/clang 180 
pip3 install distro
```

change the `carla/Util/BuildTools/Setup.sh:73` to:
```
LLVM_BASENAME=llvm-12.0
```

and replace `python` with `python3` in `carla/Util/BuildTools/BuildCarlaUE4.sh:134` and `carla/Util/BuildTools/BuildCarlaUE4.sh:137`

closed 2023-06-08 by buechmarc

sashankneupane, 2023-06-16:
I followed @dburakovtechsolutions instructions and `make PythonAPI` and `make launch` worked fine, but when I tried to run an example from PythonAPI, it gives this error:

```
File "/home/sn3006/carla/PythonAPI/examples/generate_traffic.py", line 24, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'
```

I am very new to using UnrealEngine and Carla and I might have missed some very simple steps, so if you could tell/guess what might have gone work, it would be wonderful. Thank you very much!

murthybalaji0725, 2023-06-19:
@sashankneupane Check the official documentation. It is a common issue with the path variable. Based on your carla root location and carla egg file location, you need to make changes to the path variable. You can make the changes in bashrc as well.

sashankneupane, 2023-07-15:
thank you for your comment @murthybalaji0725 . I stopped working on CARLA for a while, but I was able to use your suggestion to make it work. I was also able to make it work using docker.

murthybalaji0725, 2023-07-20:
> > thank you for your comment @murthybalaji0725 . I stopped working on CARLA for a while, but I was able to use your suggestion to make it work. I was also able to make it work using docker.
> 
> Can you show me the dockerfile with which you build CARLA 0.9.14 on Ubuntu 22.04 ?

I didnt use a docker file. Also i havent ever worked with them. I simply followed the linux build from source. And I believe I was working with Ubuntu 18.04 given its better compatbility

sashankneupane, 2023-07-20:
@TUMGEKA I do not know much about docker and I used it only to get CARLA up and running, so I only pulled docker image carlasim/carla:latest and ran it. You can see the scripts I used to run it here: https://github.com/sashankneupane/carla-installation/

roshambo919, 2023-10-06:
xerces 3.2.4 will do it