carla-simulator/carla #6397: Python API misses agents

gabrik, 2023-04-11:
Hi all,

I'm using CARLA client API within a docker container and I need to access to the GlobalRoutePlanned that is part of the agents module but It seems not installed.

CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04
Problem you have experienced: When installing python API from pip the agents modules is missing
What you expected to happen: The agents modules to be present
Steps to reproduce: 
```
pip3 install carla==0.9.14
python3 -c "from carla.agents.navigation.global_route_planner import GlobalRoutePlanner"
```

Any suggestion?

DeepDuke, 2024-01-21:
Try this to copy the `agents` package into your python's site-packages folder. If your python version is `python3.8`, run:
```bash
cp -r PythonAPI/carla/agents .local/lib/python3.8/site-packages/

```

DeepDuke, 2024-01-21:
Weirdly, the developers didn't put `agents` as a subpackage of Carla client `carla` but as an individual package.

firemark, 2024-02-16:
Is there any reason for that? Coping from the build or the docker image is annoying.

If not, I will prepare pull request with updated setup.py in the next week.