carla-simulator/carla #6401: I am getting the same error as below

spulivar, 2023-04-14:
I am getting the same error as below 

git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26
Cloning into 'C:/MinGW/msys/1.0/home/pamid/UnrealEngine_4.26'...
info: please complete authentication in your browser...
remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

Kindly provide the access to the repository and for the special privilege do we need to raise any ticket in the git hub portal as customer ticket.

Kindly advice the same for the error ?

_Originally posted by @spulivar in https://github.com/carla-simulator/carla/issues/4802#issuecomment-1509282158_

zeidk, 2023-06-10:
Please, we need an update about this.

zeidk, 2023-06-20:
Everything will work after you accept an email invitation to the Epic Games repo.

alahi45, 2023-10-02:
Hi, I'm also facing the same issue. I have linked Epic Games with Github account. But I didn't get any invitation to join the Repo. Nothing is showing in Github>Your Organization. How much time they take to send invitation for joining the Repo?