carla-simulator/carla #6402: Can't use trafic manager with my custom map

BehemothTB, 2023-04-15:
CARLA version: 0.9.14 build from source
Platform: ubuntu 20.04.6 LTS
When I start manual_control.py with my custom map imported from RoadRunner, I get the following output:
```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.z

    W            : throttle
    S            : brake
    A/D          : steer left/right
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down
    CTRL + W     : toggle constant velocity mode at 60 km/h

    L            : toggle next light type
    SHIFT + L    : toggle high beam
    Z/X          : toggle right/left blinker
    I            : toggle interior light

    TAB          : change sensor position
    ` or N       : next sensor
    [1-9]        : change to sensor [1-9]
    G            : toggle radar visualization
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    O            : open/close all doors of vehicle
    T            : toggle vehicle's telemetry

    V            : Select next map layer (Shift+V reverse)
    B            : Load current selected map layer (Shift+B to unload)

    R            : toggle recording images to disk

    CTRL + R     : toggle recording of simulation (replacing any previous)
    CTRL + P     : start replaying last recorded simulation
    CTRL + +     : increments the start time of the replay by 1 second (+SHIFT = 10 seconds)
    CTRL + -     : decrements the start time of the replay by 1 second (+SHIFT = 10 seconds)

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit

Fatal Python error: PyEval_SaveThread: NULL tstate
Python runtime state: initialized

Current thread 0x00007f5c34407740 (most recent call first):
  File "manual_control.py", line 382 in __init__
  File "manual_control.py", line 1270 in game_loop
  File "manual_control.py", line 1372 in main
  File "manual_control.py", line 1380 in <module>
已放弃 (核心已转储)
```
In line 382, the program try to set autopilot to true. If I comment this line, it works. But if I use synchronous mode, it crashes again. This time, the problem caused by the function`get_trafficmanager()`. It seems that any function relative to traffic manager doesn't work with my map.
Additionally, Town 01, 02, etc. which are from official assets works well. And when I copy my map to my colleague's computer it works well too. My colleague has the same CARLA version and ubuntu version. Is that any of my setting is wrong?

jinwookim97, 2023-05-11:
The same problem happened to me. Is it resolved?

BehemothTB, 2023-05-11:
已收到

jinwookim97, 2023-05-11:
Can you tell me how it worked out?

yu-ju-chiu, 2023-05-18:
Can you run generate_traffic.py in RR scene?
The CarlaUE4.exe always gets stuck, when I run python file.
Anyone can help me?

BehemothTB, 2023-05-30:
> 

No, I didn't solve it, sorry. 
But I have a guess. I have another custom map which is smaller. This smaller map works correctly. So, perhaps it's just that my poor computer is unable to run the program with so large a map? My CPU is i5-9400. My graphic card is a 4GB RAM 1050Ti. And my computer RAM is 8GB. I hope this information will help you.

BehemothTB, 2023-05-30:
> Can you tell me how it worked out?

No, I didn't solve it, sorry.
But I have a guess. I have another custom map which is smaller. This smaller map works correctly. So, perhaps it's just that my poor computer is unable to run the program with so large a map? My CPU is i5-9400. My graphic card is a 4GB RAM 1050Ti. And my computer RAM is 8GB. I hope this information will help you.

BehemothTB, 2023-05-30:
> Can you run generate_traffic.py in RR scene? The CarlaUE4.exe always gets stuck, when I run python file. Anyone can help me?

I can't run it with my custom map, because it uses traffic manager. But it works correctly with any one of official maps.