carla-simulator/carla #6403: How to apply texture on traffic signs?

pallavi-garg, 2023-04-15:
Hi,

I am working on an autonomous driving vehicle using PythonAPI. For some test cases, I need to add texture over the traffic signs like (traffic lights, stop signs, etc.), but I am able to apply texture only to buildings. Could someone please help me in modifying the appearance of actors in map using PythonAPI ( i.e., without modifying the map or props)?

Thanks.

neilsambhu, 2023-04-18:
@pallavi-garg, I have encountered the same issue with only being able to modify the appearance of a building following https://carla.readthedocs.io/en/latest/tuto_G_texture_streaming/ . I am unable to modify the appearance of vehicles. Are you using the apply_color_texture_to_object(...) method? I modified TGA files exported from Unreal Engine compiled from source. I suspect there is an issue with the UV mapping not being preserved when I export the modified file from GIMP. I think that the building has a simpler layout than a vehicle. Which software did you use to modify the traffic signs?

daohu527, 2024-11-08:
any update? https://github.com/carla-simulator/carla/issues/5748