carla-simulator/carla #6408: Radar sensor output

lucalazzaroni, 2023-04-17:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04

I am using Carla for a Reinforcement Learning-based project. I attached a radar sensor to the ego car but, since I have to feed the radar output data to a neural network, I do not understand how I should process the data. I would like to be able to set the azimuth and altitude resolutions but it seems to be not possible. The only parameter that i found useful is the `points_per_second` one, but I do not understand the logic by which points are generated. How many lasers are used vertically and horizontally? If, for instance, I set the `points_per_second` variable to 100, then I keep all elements in the scene stationary, why the generated points are always different second by second?

Thanks in advance for your help.

JosenJiang, 2023-04-25:
The radar in Carla will randomly cast rays at the range of horizontal_fov and vertical_fov you set.

lucalazzaroni, 2023-04-25:
Thank you @JosenJiang for your kind response. Do you know if there’s a way to avoid randomness in rays casting? So for instance to cast by row or by column

JosenJiang, 2023-04-25:
Modify Radar.cpp refer to RayCastLidar.cpp @lucalazzaroni

closed 2023-04-26 by lucalazzaroni