carla-simulator/carla #6410: Incorrect Position of 3d Bounding Boxes

RostyslavUA, 2023-04-18:
I am trying to visualize 3d bounding boxes using open3d. The bounding boxes show up, but in the wrong place. In the video, the bounding boxes (shown in red) show up way above the vehicle.

https://user-images.githubusercontent.com/49762976/232761382-45169887-6bb7-4a53-a607-8f31be4c096f.mp4

My lidar callback function looks as follows
```
def lidar_callback(point_cloud, point_list, world, vehicle, lidar):
    """Prepares a point cloud with intensity
    colors ready to be consumed by Open3D"""
    data = np.copy(np.frombuffer(point_cloud.raw_data, dtype=np.dtype('f4')))
    data = np.reshape(data, (int(data.shape[0] / 4), 4))

    # Isolate the intensity and compute a color for it
    intensity = data[:, -1]
    intensity_col = 1.0 - np.log(intensity) / np.log(np.exp(-0.004 * 100))
    int_color = np.c_[
        np.interp(intensity_col, VID_RANGE, VIRIDIS[:, 0]),
        np.interp(intensity_col, VID_RANGE, VIRIDIS[:, 1]),
        np.interp(intensity_col, VID_RANGE, VIRIDIS[:, 2])]

    # Isolate the 3D data
    points = data[:, :-1]

    # We're negating the y to correclty visualize a world that matches
    # what we see in Unreal since Open3D uses a right-handed coordinate system
    points[:, :1] = -points[:, :1]

    int_color *= np.array([0, 255, 0])
    world_2_lidar = lidar.get_transform().get_inverse_matrix()
    for npc in list(world.get_actors().filter('*vehicle*')) + list(world.get_actors().filter('*pedestrian*')):
    # Filter out the ego vehicle
        if npc.id != vehicle.id:
            bb = npc.bounding_box
            dist = npc.get_transform().location.distance(vehicle.get_transform().location)
            # Filter for the vehicles within 50m
            if dist < 50:

            # Calculate the dot product between the forward vector
            # of the vehicle and the vector between the vehicle
            # and the other vehicle. We threshold this dot product
            # to limit to drawing bounding boxes IN FRONT OF THE CAMERA
                forward_vec = vehicle.get_transform().get_forward_vector()
                ray = npc.get_transform().location - vehicle.get_transform().location
                verts = [[] for _ in range(3)]
                if forward_vec.dot(ray) > 1 or True:
                    verts = [v for v in bb.get_world_vertices(npc.get_transform())]
                    verts = [[v.x, v.y, v.z] for v in verts]
                verts = np.array(verts)
                if verts.shape[1] > 0:
                    colors = np.ones((verts.shape[0], 3))*np.array([255, 0, 0])
                    points = np.concatenate((points, verts), 0)
                    int_color = np.concatenate((int_color, colors), 0)
    point_list.points = o3d.utility.Vector3dVector(points)
    point_list.colors = o3d.utility.Vector3dVector(int_color)
```

Do you know why does it happen and how can I circumvent the problem?

CARLA version: 0.9.14
Platform/OS: Ubuntu 20.02

RostyslavUA, 2023-04-26:
@JosenJiang Thank you for your reply. I have made sure that the frame numbers are the same by adding the following line to my `lidar_callback`:
```
print(f"\nworld.get_snapshot().frame: {world.get_snapshot().frame}\tpoint_cloud.frame: {point_cloud.frame}")
```
And this is the result that I've got
```
world.get_snapshot().frame: 82289	point_cloud.frame: 82289
world.get_snapshot().frame: 82290	point_cloud.frame: 82290
world.get_snapshot().frame: 82291	point_cloud.frame: 82291
world.get_snapshot().frame: 82292	point_cloud.frame: 82292
world.get_snapshot().frame: 82293	point_cloud.frame: 82293
world.get_snapshot().frame: 82294	point_cloud.frame: 82294
world.get_snapshot().frame: 82295	point_cloud.frame: 82295
world.get_snapshot().frame: 82296	point_cloud.frame: 82296
world.get_snapshot().frame: 82297	point_cloud.frame: 82297
world.get_snapshot().frame: 82298	point_cloud.frame: 82298
```
So they are indeed the same. 
I believe that the reason is rather in incorrect geometric transformation: as you can see in the video, the lidar data and the bounding boxes are synchronized, but they show up in the different places.
Do you know how to fix that?

Tottowich, 2023-06-25:
@RostyslavUA I believe that the transformation is incorrect. I've used the following code to create the 8-point bounding box visualization.

```python
assert self.sensor is not None, f"{self.name} has not been added to a sensor yet"
        location = self.sensor.location()
        forward_direction = self.sensor.forward_vector(normalized=True)
        w2c = self.sensor.w2s()[0:4, 0:4]
        for bbox in self.bounding_box_set:
            # Filter bounding boxes that are too far away
            if bbox.location.distance(location) > self.sensor_range:
                continue
            ray = bbox.location - location
            ray_direction = ray/ray.length()

            # Filter bounding boxes that are not in the field of view
            cos_sim = forward_direction.dot(ray_direction)
            if cos_sim < self.cos_threshold:
                continue
            verts = np.array([carla_vec2np_array(v) for v in bbox.get_world_vertices(carla.Transform())])
            verts = np.hstack((verts, np.ones((verts.shape[0], 1))))
            verts = np.dot(w2c, verts.T).T
            # Negate the y-axis to match the camera coordinate frame
            verts[:, 1] *= -1
            # Rescale the points to the world coordinate frame
            verts[:, 0] /= verts[:, 3]
            verts[:, 1] /= verts[:, 3]
            verts[:, 2] /= verts[:, 3]
            # Remove the fourth component
            verts = verts[:, 0:3]

            scene.points.extend(o3d.utility.Vector3dVector(verts))
            scene.colors.extend(o3d.utility.Vector3dVector(np.tile([255, 0, 0], (verts.shape[0], 1))))
        self.sensor.pcd = scene
        return scene
        
def w2s(self): # function in my sensor wrapper class.
        """
        Get the world-to-sensor transformation matrix.

        Returns
        -------
        np.ndarray
            The world-to-sensor transformation matrix.
        """
        return np.array(self.transform().get_inverse_matrix())
```

alex-920, 2024-12-10:
@RostyslavUA have you resolved the issue yet? I tried integrating the code provided by @Tottowich, but unfortunately it doesn't work for me.