carla-simulator/carla #6414: CARLA 0.9.14 - PythonAPI - ModuleNotFoundError: No module named 'carla.libcarla.command'

gabrik, 2023-04-19:
Hi,

While writing our own setup script to spawn cars and pedestrians in the simulator, we encountered this issue when trying to import the `SpawnActor`, or any module under `carla.command`

Our setup script is based on the `/carla/PythonAPI/examples/generate_traffic.py` one.

Any suggestion?

CARLA version:  0.9.14
Platform/OS: Ubuntu 20.04
Problem you have experienced: ModuleNotFoundError: No module named 'carla.libcarla.command'
What you expected to happen: to import the module
Steps to reproduce: 

Install carla python from pypi: and try to import a command

```
pip3 install carla==0.9.14
python3
> from carla.command import SpawnActor
```

vagaiprem, 2023-08-14:
i am also facing same error plz help anyone in windows environment

gabrik, 2023-09-07:
> Solution is here https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/generate_traffic.py
> 
> ```
> import carla
> SpawnActor = carla.command.SpawnActor
> ```

That does not work if CARLA python API is installed from pip.

Youngzipper, 2023-10-15:
> i am also facing same error plz help anyone in windows environment

have you found the solution? i encountered the same problem in windows.

Daraan, 2024-09-18:
I've now proposed a working fix for this see: https://github.com/carla-simulator/carla/pull/8161

PR #8161 (2025-01-17): Correct "libcarla.command" namespace and allow `from carla.command import ...`
Fixes  #6414  

Currently `import carla.command` or `from carla.command import ...` is not possible.

This will raise:
```python
# command.py
      7 # pylint: disable=W0401
----> 8 from .libcarla.command import *

ModuleNotFoundError: No module named 'carla.libcarla.command'
```

This is because the `command` module is registered as `libcarla.command` in `sys.modules`. This is not really correct.

---

This PR adjusts this:
- renames the module to `carla.libcarla.command` 
- provides a duplicate `carla.command` tha [... cut]
commit aeeff990 (2025-01-17): Correct "libcarla.command" namespace and allow `from carla.command import ...` (#8161)

* Correct "carla[.libcarla].command" in sys.modules

Fixes  #6414  https://github.com/carla-simulator/carla/issues/6414

* Updated changelog