carla-simulator/carla #6419: make LibCarla failing  for 0.9.13 & 0.9.14 versions

vrinda14, 2023-04-21:
CARLA version: 0.9.13 & 0.9.14
Platform/OS:Ubuntu
Problem you have experienced: unable to build LibCarla
Steps to reproduce: Using a Dockerfile to build carla. 
Trying to build Carla but failing in make LibCarla command. Tried with both 0.9.13 as well as 14 versions and tried increasing the memory, reinstalling packages, removing clang8, lldb packages. 
Unable to proceed further.
Any suggestions to solve the issue ? 
Thanks in advance
 In Carla 0.9.13 :

[52/54] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/road/Map.cpp.o18:18:24  
ninja: build stopped: subcommand failed.18:18:24  make: *** [Util/BuildTools/Linux.mk:118: LibCarla.server.release] Error 1

In Carla 0.9.14 :

Building CXX object LibCarla/cmake/client/CMakeFiles/carla_client.dir/__/__/source/carla/road/Map.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Util/BuildTools/Linux.mk:124: LibCarla.client.release] Error 1

MichelleSchaaf, 2023-08-15:
I have the same issue. @vrinda14 have you solved this problem?

forlovedriveless, 2024-05-31:
have you solved this? @vrinda14