carla-simulator/carla #6436: he system cannot find the path specified: 'dependencies/lib'

sharikalog7, 2023-05-02:
error: subprocess-exited-with-error
  
  python setup.py egg_info did not run successfully.
  exit code: 1
  
  [8 lines of output]
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "C:\Users\ndalien2\AppData\Local\Temp\pip-install-ttxbprrd\carla_2381ee75c72943788ce6181b22b60c5c\setup.py", line 114, in <module>
      ext_modules=get_libcarla_extensions(),
    File "C:\Users\ndalien2\AppData\Local\Temp\pip-install-ttxbprrd\carla_2381ee75c72943788ce6181b22b60c5c\setup.py", line 75, in get_libcarla_extensions
      libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
  FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.

See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

[notice] A new release of pip available: 22.3.1 -> 23.1.2
[notice] To update, run: python.exe -m pip install --upgrade pip

NnaYelsel, 2023-05-02:
Hi, 
Could you give more details about your configuration etc... ? This message is rough.

I have the same problem when I try to compiler CARLA v0.9.14 on windows, it's not creating the path. 
Nevertheless you can create dependencies/lib manually in PythonAPI/carla and see if that's work. 

For me that's not working because further, I have error because some folder named "client" and "server" are not created either. 

Does someone in the Carla team knows which bat file is dedicated to the creation of these folders ? 

Best regards.

StevenCoral, 2023-05-08:
Getting the exact same error. It seems that pip is trying to download carla version 0.9.5, while 0.9.14 does not exist (even though Pypi shows 0.9.14 as the most recent version.

debipe20, 2023-06-13:
I downgraded to Python3.8.10. But even after I am getting error "FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'" while building CARLA0.9.13.

GraySnail, 2023-09-08:
If you install carla using package but not source compile on Windows, python 3.7 is required. Other python is not supported.

AhmedOmarSallam, 2023-10-22:
> Getting the exact same error. It seems that pip is trying to download carla version 0.9.5, while 0.9.14 does not exist (even though Pypi shows 0.9.14 as the most recent version). Looking at the downloadable files, is it possible carla client library is only supported for up to Python 3.8? it does state so in the quick installation guide - perhaps Devs should add this point to the Windows Build Minor Installations section?
> 
> Edit: I can confirm that downgrading Python to 3.8 solved the issue. Please add this important detail to the documentation.

That works for me, Thanks.

shibbir, 2025-02-11:
I'm still facing issues. I have Python v3.12.9 installed.

[image]