carla-simulator/carla #6444: Collision and LaneInvasion sensor does not work properly during replay

Morphlng, 2023-05-04:
- CARLA version: 0.9.13

- Platform/OS: Windows/Ubuntu

- Problem you have experienced: 
   I'm trying to collect data for RL training from a list of replay files. I'm setting the world to synchronous mode once the replay is fully loaded. I'll then spawn necessary sensors to collect the data for RL algorithm, such as RGB image and semantic information. The problem right now is that the Collision Sensor and LaneInvasion Sensor does not call the callback function at the point where the event happened. Instead, the result of the callback usually returns after the replay is ended (this is not always, if the replay is really long, it might return in later time). The RGB sensor is doing simply fine.

  [image]

- What you expected to happen:
   The sensor should call the callback function at once when a corresponding event happens.

- Steps to reproduce:
   I'll try to provide a minimal reproducible code when I have time, right now I can only provide you with the code to retrieve vehicle actors, you can attach sensor to actors from here:

```python
    def parse_replay_file(self, filepath):
        '''Parse actor information from a replay file

        Args:
            filepath (str): The path to the replay file.
        
        Returns:
            actors (list[dict]): A list of actor information.
            total_duration (int): The total duration (in seconds).
        '''
        if self._client is not None:
            info = self._client.show_recorder_file_info(filepath, False)

            # Parse actor information
            actor_raw_info = [line.strip() for line in info.splitlines()]
            total_duration = float(actor_raw_info[-1].split(' ')[1].strip())
            pattern = r'Create (\d+): ([\w.]+) \(\d+\) at \((\-?\d+(?:\.\d+)?), (\-?\d+(?:\.\d+)?), (\-?\d+(?:\.\d+)?)\)'
            re_match = [match for match in [re.match(pattern, line) for line in actor_raw_info] if match is not None]

            actors = [{
                'id': int(actor_info[0]),
                'blueprint': actor_info[1],
                'location': (float(actor_info[2]), float(actor_info[3]), float(actor_info[4]))
                } for actor_info in [match.groups() for match in re_match] if 'vehicle' in actor_info[1].lower()]

            return actors, total_duration

        raise Exception("Client is not connected")  

    def replay_file(self, filepath, start=0, duration=0, follow_vehicle=0):
        """Start to replay a file in sync mode, tick till vehicle is spawned.
        
        Args:
            filepath (str): The path to the file.
            start (int): The start frame. Negative values means from the end. 0 means the beginning.
            duration (int): The duration of the replay. 0 means the whole file.
            follow_vehicle (int): The id of the vehicle to follow. 0 means no follow.

        Returns:
            list[carla.Actor], list[dict]: A list of actors and a list of actor information.
            int: The total duration (in seconds).
        """

        if self._client is not None:
            actors_info, total_duration = self.parse_replay_file(filepath)

            # Replay
            self.toggle_sync_mode(True)
            self._client.replay_file(filepath, start, duration, follow_vehicle)

            # Reset DataProvider
            tm_port = self._data_provider.get_traffic_manager_port()
            self._data_provider.reset(completely=True)
            self._data_provider.set_client(self._client)
            self._data_provider.set_world(self._client.get_world())
            self._data_provider.set_traffic_manager_port(tm_port)

            # Tick till vehicle is spawned
            while True:
                self.tick()
                actors = [self.get_actor_from_server(info["id"]) for info in actors_info]
                if all([actor is not None for actor in actors]):
                    break
            
            return actors, actors_info, total_duration

        raise Exception("Client is not connected")
```

Morphlng, 2023-06-28:
It seems like if both the record and replay is in synchronized mode, this wouldn't be a problem. But I also found that it occasionally happened that the actor id in the record does not match when actual replaying.

closed 2023-06-28 by Morphlng

AOOOOOA, 2023-12-24:
> It seems like if both the record and replay is in synchronized mode, this wouldn't be a problem. But I also found that it occasionally happened that the actor id in the record does not match when actual replaying.

Hi, does the "sync mode in record" mean the Carla world is in sync mode while calling the recorder? Is there any method to check if the log file is recorded under the sync mode if we only have the recorded log file?

Morphlng, 2023-12-26:
> Hi, does the "sync mode in record" mean the Carla world is in sync mode while calling the recorder? Is there any method to check if the log file is recorded under the sync mode if we only have the recorded log file?

Synchronous mode is usually used in combination of "fixed_delta_seconds" (If not, Carla will warn you). So the simplest way to check whether the replay file is recorded under sync mode, is to check the frame time between each two frame, which should be constant.

```python
import carla

client = carla.Client("localhost", 2000)
filepath = "path_to_replay"
with open("replay_info.txt", "w") as f:
    f.write(client.show_recorder_file_info(filepath, True))
```