carla-simulator/carla #6447: CARLA 0.8.4 Windows Download File doe not exist (anymore)

Pointifix, 2023-05-04:
The download link for CARLA 0.8.4 Windows is broken:
https://github.com/carla-simulator/carla/releases/tag/0.8.4
https://drive.google.com/file/d/1CSdQAP2FwvEqQE5Opaq-tVReeW991bR2/view?usp=sharing

Is there a updated link somewhere, can someone post it or fix the link in the release?

[image]

Blyron, 2024-05-03:
We do not support 0.8.4 anymore. If you want you could clone 0.8.4 tag repo and try to build yourself, but as I say we do not support that anymore

Blyron, 2024-05-03:
Are you happy to close this issue?

Pointifix, 2024-05-03:
I "solved" the problem by switching to Ubuntu. Thank you for the reply though!

closed 2024-05-03 by Pointifix