carla-simulator/carla #6452: Carla 0.9.14 image aspect ratio has to be fixed value?

ZijianHan, 2023-05-07:
### Discussed in https://github.com/carla-simulator/carla/discussions/6451

<div type='discussions-op-text'>

<sup>Originally posted by **ZijianHan** May  7, 2023</sup>
Hi, I'm wonder does the camer image size has to follow a specific aspect ratio of 16:9 or 4:3 in carla?

I'm trying to set some non-regular to the image size, for exaple 3832*2134, or some smaller value that is not exactly 16:9, then I got some strange image output as seen in the attachment. 

Here is part of my code for setting up the camera
`          camera_bp = bp_lib.find('sensor.camera.rgb')
          camera_bp.set_attribute("image_size_x",str(3832))
          camera_bp.set_attribute("image_size_y",str(2134))
          camera_bp.set_attribute("fov",str(105))
          camera_init_trans = carla.Transform(carla.Location(z=2))
          camera = world.spawn_actor(camera_bp, camera_init_trans, attach_to=vehicle) `

Is there any way I can make the image output with any aspect ratio?
[image]
</div>

MoxDry, 2023-05-08:
If you are modifying one of the examples, you can try to modify the image size through 
`    argparser.add_argument(
        '--res',
        metavar='WIDTHxHEIGHT',
        default='1280x720',
        help='window resolution (default: 1280x720)')`
It works for me with a resolution of '5760x1080'. You may need to change the FOV for a suitable view.

ZijianHan, 2023-05-08:
> If you are modifying one of the examples, you can try to modify the image size through ` argparser.add_argument( '--res', metavar='WIDTHxHEIGHT', default='1280x720', help='window resolution [...]

Hi, thanks for the reply! Did you mean to modify the simulator window size (in that case the camera is the whole window)?

what if I have multiple cameras atached to ego vehicle with different image size, in that case I think I need to change the camera attribute right? Or did I misunderstand here, as I do not know if camera has an attribute for "--res"?

flok, 2023-05-08:
this look like the stride problem that's currently present for Windows in the 0.9.14 release. See #6085 and cba2231977d24ee78dfc8b28a77fb4528d9559b3 for the fix

ZijianHan, 2023-05-08:
Thanks a lot! I've confimed it's a bug on 0.9.14 on Windows. Works for me on Linux.