carla-simulator/carla #6465: Windows: How do I generate a Debug build of the CARLA client?

hahne, 2023-05-10:
Hello,

I am using Microsoft Visual Studio 2019 and I successfully build CARLA on Windows

I have a C++ application which uses the carla client API to receive information of a running carla simulation.
Everything works fine in Release. However, I can not build the application in debug as I do not have the debug libs of carla and the dependencies of carla.

If I try to link the release version of the libs, I get linking errors like:
`error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2'`

My question is, is there an easy way to build at least the carla client lib in debug under Windows?

hahne, 2023-05-10:
It works actually if you manually compile the debug versions of

- carla client
- boost
- RPC
- RecastNavigation

closed 2023-05-10 by hahne

ushio2021, 2023-05-30:
@hahne Hi, I wonder if you have tried to debug the associated C++ code while running a Python client? I still cannot debug the client code after manually compiling the debug versions of all associated libs, also VS2019 on Windows. Thanks in advance.

hahne, 2023-05-30:
@ushio2021 I do have a pure C++ client. You will need to compile the PythonAPI with the Debug libraries and then attach a C++ debugger on your Python process.
Something like that might work:
https://marketplace.visualstudio.com/items?itemName=benjamin-simmonds.pythoncpp-debug
But I never tried that.

ushio2021, 2023-05-30:
> @ushio2021 I do have a pure C++ client. You will need to compile the PythonAPI with the Debug libraries and then attach a C++ debugger on your Python process. Something like that might work: [...]

Thanks for your quick reply.

Actually I have built all libs (CMake Debug config) and modify the setup.py to solve some '0' and '2' errors, then tried to attach Jetbrains Rider's bundled LLDB on the Python process but always get a hint like "No executable code is associated with this line" for the breakpoint no matter where I set it.

hahne, 2023-05-30:
@ushio2021 ok sorry, no idea if this should work. I haven't tried that before.

ushio2021, 2023-05-30:
> @ushio2021 ok sorry, no idea if this should work. I haven't tried that before.

Never mind, thanks again. 

Have seen success on Linux here. Just hope someone could solve this on Windows.