carla-simulator/carla #6466: manual_control.py is not running on the carla 0.9.14

jinwookim97, 2023-05-11:
- OS: [Windows 10]
- CARLA Version [CARLA 0.9.14]
- Python version [3.8]

My private custom map was being applied.
However, manaul_control after upgrading to 0.9.14. There was a phenomenon that the py was not running(it ran normally on 0.9.11).
I was looking for a solution, and I found the keyboard control class in manual_control. world.player.set_autopilot(self._autopilot_enabled) line was annotated and executed. it's sucessful excuted. 
However, the autopilot function was not available. 
I'm trying to solve it, but I can't find a solution to this issue. Could you help me solve it. 
The output when not annotated (not executed) is as follows. 

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.14
WARNING: Simulator API version  = 0.9.14-4-gf14acb257
WARNING: No InMemoryMap cache found. Setting up local map. This may take a while... 
```

yu-ju-chiu, 2023-05-18:
Can you run others py file.
My CarlaUE4 always gets stuck, when I run python generate_traffic.py.
Anyone can help me?

AndritMuca, 2023-05-20:
Do you start the simulator with: make launch?

jinwookim97, 2023-05-22:
> 
yah,,,,

jinwookim97, 2023-05-22:
> 다른 사람의 py 파일을 실행할 수 있습니까? 내 CarlaUE4는 python generate_traffic.py를 실행할 때 항상 멈춥니다. 누구든지 나를 도울 수 있습니까?

I think, as the carla is updated to 0.9.14, the criteria for checking custom maps seem to have become stricter

jinwookim97, 2023-05-22:
> 다음 명령으로 시뮬레이터를 시작합니까? make launch?

 I did everything right by carla reference

litianhe, 2023-06-01:
> > 다음 명령으로 시뮬레이터를 시작합니까? make launch?
> 
> I did everything right by carla reference

run `pip3 uninstall carla` and then run `pip3 install carla`

AndritMuca, 2023-06-01:
Go to Python\Lib\site-packages ande delete the Carla Folder. Then run on cmd: pip install carla==0.9.14     if you have version 14 installed. If still error then: 1. make PythonAPI at carla root and make launch to start the server if you have build carla with the Unreal Engine.

ray9583, 2023-07-28:
Thankt you for your answer, I found the problem and solved it. 
The problem was my custom road. 
In Carla 0.9.11 ver, this Xodr file works fine, but since 0.9.12 ver, it seems that the operating standards of the road have changed stickly. 
Simply explain, we solved it by adding a general road between the junction and the junction.

closed 2023-07-28 by ray9583