carla-simulator/carla #6474: make Python API error for carla0.9.13

yansx, 2023-05-16:
CARLA version:0.9.13
Platform/OS:Ubuntu18

Problem you have experienced:
When I "make PythonAPI",some error occurred:

“
BuildPythonAPI.sh: 
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing dependency_links to source/carla.egg-info/dependency_links.txt
writing top-level names to source/carla.egg-info/top_level.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
clang-8 -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -Idependencies/include -I/home/t/anaconda3/include/python3.6m -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.6/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
gcc -pthread -shared -B /home/t/anaconda3/compiler_compat -L/home/t/anaconda3/lib -Wl,-rpath=/home/t/anaconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.6/source/libcarla/libcarla.o -Ldependencies/lib -o build/lib.linux-x86_64-3.6/carla/libcarla.cpython-36m-x86_64-linux-gnu.so /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libcarla_client.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/librpc.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libboost_filesystem.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libRecast.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libDetour.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libDetourCrowd.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libosm2odr.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libxerces-c.a -lz /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libproj.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libsqlite3.a /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libboost_python36.a -lpng -ljpeg -ltiff -lstdc++
/home/t/anaconda3/compiler_compat/ld: /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libproj.a(geodesic.c.o): relocation R_X86_64_32S against `.rodata.cst8' can not be used when making a shared object; recompile with -fPIC
/home/t/anaconda3/compiler_compat/ld: /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libproj.a(wkt1_generated_parser.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/home/t/anaconda3/compiler_compat/ld: /home/t/carla-0.9.13/PythonAPI/carla/dependencies/lib/libproj.a(wkt2_generated_parser.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/home/t/anaconda3/compiler_compat/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
error: command 'gcc' failed with exit status 1
Util/BuildTools/Linux.mk:87: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

”
how can I solve the problem?

moti-milab, 2023-05-17:
Hi, in fact I'm running this today from scratch and can report that at least two libraries need to be updated during the installer process.
By editing files under _Util\InstallersWin_, I had to update zlib version to 1.2.13 (install_zlib.bat, line 51) because the previous one doesn't exist in their repository, same for xercesc (install_xercesc.bat, line 43) should be upgraded to 3.2.4.

Still in the process.

Moti.

awiswasi, 2023-07-25:
Currently have the same issue. I am looking into it.
If I find anything, I will update this thread.
In the meantime, if anyone have found a solution please let me know!

Angeladady, 2023-09-20:
-fPIC is only added for C++ files but C files.  Add CMAKE_C_FLAGS="-fPIC" to the PROJ part in the file Util/BuildTools/Setup.sh.

Veraleeee, 2023-10-01:
Could you please take a picture? I haven't solving this ,yet. especially when BuildOSM2ODR.sh

Croquembouche, 2024-01-31:
Same issue here.

Croquembouche, 2024-01-31:
@Veraleeee 

I deleted the proj-install and proj-server from the Build folder and recompiled, it now works.

GOOGOL666, 2024-06-11:
[image]

> Could you please take a picture? I haven't solving this ,yet. especially when BuildOSM2ODR.sh

M-Colley, 2026-10-26:
Draft PR #10027 targets the `libproj.a` relocation/PIC failure shown in this report by building PROJ with position-independent code on `ue4-dev`. It currently has merge conflicts. This is scoped to that linker failure; a complete Linux Python API rebuild remains unrun.