carla-simulator/carla #6476: make PythonAPI fails for windows carla 0.9.13

moti-milab, 2023-05-17:
CARLA version: 0.9.13
Platform/OS: Windows 10
Problem you have experienced: Failure of compilation due to missing packages
Steps to reproduce: issue make PythonAPI in a development terminal
Other information (documentation you consulted, workarounds you tried):

Please note that some dependent libraries are now missing online.

By editing files under _Util\InstallersWin_, I had to update zlib version to 1.2.13 (_install_zlib.bat_, line 51) because the previous ones doesn't exist in their repository, same for xercesc (_install_xercesc.bat_, line 43) should be upgraded to 3.2.4.

Still in the process.

Moti.

NnaYelsel, 2023-05-23:
Do you know which package are missing ?

moti-milab, 2023-05-25:
Hi, specifically it failed on the two I mentioned above: zlib==1.2.11 and xercesc==3.2.3.
Going through another repository (DReyeVR) seems like they can be found on a different mirror.

liujiawei725, 2023-05-30:
Try to download the zlib1213.zip file from https://www.zlib.net/, and rename the zip file as zlib-1.2.13.zip. Place the renamed the zip file in the carla/build directory and execute make PythonAPI

moti-milab, 2023-05-30:
Thanks @liujiawei725, that's exactly what I did, please also take into consideration that due to some awkward reason related to the way cmake is probably configured, it is also necessary to copy this output directory and have its name as 1.2.11 again, otherwise some further compilation/linking/resolution step fails.

moti-milab, 2023-06-09:
Hi @debipe20, not specifically, but it looks like it cannot find _make_ and the path it uses might come from some environment variable (check why it refers to _C:\GnuWin32\bin_, this should have been a reference to _make_).

regiin09, 2024-08-25:
> I am getting error on make PythonAPI. Did you face this type of issues? Could you guide me?
> 
> C:\Users\ddas\Documents\carla>make PythonAPI -[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all -[Setup]: Asynchronous jobs: 12 -[Setup]: Boost toolset: msvc-14.2 -[Setup]: Install directory: "C:\Users\ddas\Documents\carla\Build" -[Setup]: Installing zlib... -[install_zlib]: [Batch params]: --build-dir "C:\Users\ddas\Documents\carla\Build" -[install_zlib]: Not downloading zlib because already exists the folder "C:\Users\ddas\Documents\carla\Build\zlib-source". CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required): Compatibility with CMake < 2.8.12 will be removed from a future version of CMake.
> 
> Update the VERSION argument value or use a ... suffix to tell CMake that the project does not need compatibility with older versions.
> 
> -- The C compiler identification is MSVC 19.29.30148.0 -- Detecting C compiler ABI info -- Detecting C compiler ABI info - failed -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - broken CMake Error at C:/Program Files/CMake/share/cmake-3.26/Modules/CMakeTestCCompiler.cmake:67 (message): The C compiler
> 
> ```
> "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"
> ```
> 
> is not able to compile a simple test program.
> 
> It fails with the following output:
> 
> ```
> Change Dir: C:/Users/ddas/Documents/carla/Build/zlib-source/build/CMakeFiles/CMakeScratch/TryCompile-czrf08
> 
> Run Build Command(s):C:/Program Files/CMake/bin/cmake.exe -E env VERBOSE=1 nmake -f Makefile /nologo cmTC_cf1bb\fast &&     C:\GnuWin32\bin  -f CMakeFiles\cmTC_cf1bb.dir\build.make /nologo -L                  CMakeFiles\cmTC_cf1bb.dir\build
> 'C:\GnuWin32\bin' is not recognized as an internal or external command,
> operable program or batch file.
> NMAKE : fatal error U1077: 'C:\GnuWin32\bin' : return code '0x1'
> Stop.
> ```
> 
> CMake will not be able to correctly generate this project. Call Stack (most recent call first): CMakeLists.txt:4 (project)
> 
> -- Configuring incomplete, errors occurred!
> 
> ```
> -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
> -[install_zlib]: [CMAKE ERROR] Possible causes:
> -[install_zlib]:                - Make sure "CMake" is installed.
> -[install_zlib]:                - Make sure it is available on your Windows "path".
> -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
> -[install_zlib]: Exiting with error...
> ```
> 
> -[Setup]: Ok, and error ocurred, don't panic! We have different platforms where you can find some help :)
> 
> * Make sure you have read the documentation:
>   http://carla.readthedocs.io/en/latest/how_to_build_on_windows/
> * If the problem persists, you can ask on our Github's "Building on Windows" issue:
>   [Building on Windows 0.8.X #21](https://github.com/carla-simulator/carla/issues/21)
> * Or just use our Discord channel!
>   We'll be glad to help you there :)
>   https://discord.gg/42KJdRj
>   make: *** [setup] Error 1

HAVE YOU FOUND A SOLUTION FOR THIS ?