carla-simulator/carla #6477: Delayed Acceleration even with Full Throttle

SailorBrandon, 2023-05-17:
CARLA version: 0.9.10.1
Platform/OS: ubuntu 20.04

### Description:
I am currently utilizing the Tesla Model 3 blueprint in Carla and have encountered an issue related to delayed acceleration. Despite setting the throttle to its maximum value of 1 (full), there is a noticeable delay of approximately 2 seconds before the car starts accelerating.

### Steps to Reproduce:
1. Launch Carla with the Tesla Model 3 blueprint.
2. Set the throttle input to its maximum value of 1 (full).
3. Observe the car's behavior and note the delay before acceleration kicks in.

### Expected Behavior:
The Tesla Model 3 blueprint should initiate acceleration promptly upon receiving a throttle input of 1 (full).

### Actual Behavior:
The car experiences a delay of approximately 2 seconds before it starts accelerating, despite the throttle being set to its maximum value.

ivevasiljevic, 2023-05-23:
From my experience, I noticed that the moment a vehicle spawns (I have mostly used Tesla Model 3) it's like the vehicle is stuck to the ground, and it takes some time for a vehicle to get “unstuck”. Once it does, then it's much more responsive to the acceleration. I guess it's just a nature of Carla.

CodeTornato, 2023-06-09:
@ivevasiljevic  @SailorBrandon  Hey, I'm experiencing this problem too(I'm using carla 0.9.13), I found a solution here:
https://github.com/carla-simulator/carla/issues/1640
but it did not work me, still getting half second of delay no matter what
[image]
let me know if you guys find a way to cancel that delay

kaynaremre, 2023-06-12:
Hi Guys, I actually got the same issue with most of the vehicles in the blueprint library. 

First workaround I used was choosing the first gear with apply_control(). That decreased the response time, but still there was delay.

It was quiet obvious that the vehicle dynamics are not simulated correctly. So I have changed Physics substepping settings as increase # max_substeps and decrease max_substep_delta_time.

The code below shows my current settings for physics substepping and I can see that the vehicle dynamics are simulated more precisely, and the delay was gone.

```
settings = world.get_settings()
settings.substepping = True
settings.max_substep_delta_time = 0.001
settings.max_substeps = 100
world.apply_settings(settings)
```
Hope this helps :)

kaynaremre, 2023-06-14:
@CodeTornato Are you running on synchronous mode and fixed time step ?

CodeTornato, 2023-06-14:
@kaynaremre well i did not set it to synchronous mode explicitly, I'm just knowing it right know after you told me it can be run on different mode

VidaAPI, 2023-08-03:
@CodeTornato Hi, can you please explain how did you get the gradual increase in the speed? I'm also using Tesla3. But i cant control the acceleration, it accelerates with the max throttle, and decelerates with the max too.
This can help me so much, if you share.

CodeTornato, 2023-08-04:
@VidaAPI hey,in my case,I have a planning and control algorithms, so control decide how much throttle it should apply base on the desired speed. 
and can you provide me with your source code so I could take a look on why it's being decelerating.
And Also By The Way,I have updated my carla to the newest version(0.9.14), because in newest version there is a new api for giving a ackermman control,so you could control speed and acceleration directly,cuz using throttle is hard to reach a given speed.
[image]

ChaoTan201, 2024-04-21:
Hi, I encountered the same problem, Could you give me some advices? Thx : )